Duet3D/RepRapFirmware #557: 3.3.0 with Maestro -  generic.heaterStates.disconnected

tonypartridge, 2022-02-07:
Anyone have any hints for this issue? Just upgraded and got the attached, latest build from the configuration tool and tried manually installed.
[image]
[image]

jaysuk, 2022-02-07:
you're running an older version of the firmware. 2.05 with a newer version of DWC. please upgrade the firmware version on the maestro to 3.3.0

tonypartridge, 2022-02-07:
Thanks, I have tried but it stays at 2.5 🤯

jaysuk, 2022-02-07:
Upload these 3 zip files, one at a time in the system tab. Don't extract them first. Reboot after each. Use M115 in the gcode console to verify the firmware has been applied.
https://github.com/Duet3D/RepRapFirmware/releases/download/2.05.1/Duet2Firmware-2.05.1.zip
https://github.com/Duet3D/RepRapFirmware/releases/download/3.0/Duet2and3Firmware-3.0.zip
https://github.com/Duet3D/RepRapFirmware/releases/download/3.3/Duet2and3Firmware-3.3.zip
That will get your firmware and DWC up to date.

You can see the change logs here:
https://github.com/Duet3D/RepRapFirmware/wiki/Changelog-RRF-3.x

For your config, might be a good idea to run through the configurator tool and generate a fresh set for RRF3.
https://configtool.reprapfirmware.org/Start

Backup your existing config files in the sys folder in case you want to switch back to RRF2. It’s easy to switch back and forth, just upload the zip file for the version you want and then upload your config files.

These documents will come in handy during the conversion.
https://duet3d.dozuki.com/Wiki/RepRapFirmware_3_overview
https://duet3d.dozuki.com/Wiki/Gcode

tonypartridge, 2022-02-07:
Thanks @jaysuk makes sense, I didn't appreciate we had to do the 3 jumps. All sorted now. 

It would be good for the webui to check firmware version and implement a popup advising the above 3 need installing in order. 

I had to also manually move Duet2_SDiap32_Maestro.bin to the firmware folder for it to finish applying the update from the configuration build.

closed 2022-02-07 by tonypartridge