Duet3D/RepRapFirmware #558: [FeatureRequest]:  M282 - Detach Servo to save the life of servos

disneytoy, 2022-02-13:
### Is your feature request related to a problem? Please describe.

M282 - Detach Servo

"Detach a servo, which turns off its power. The servo will be attached (powered up) before its next move."

I use inexpensive hobby servos in a pen plotter.  A plot can take many hours, but the servo engages very briefly.  These servos tend to burn out or overheat.  If M282 was implemented, it would greatly increase the life of the servos, by de-energizing them until they are envoked again with M280.

Thanks!

Max

### Describe the solution you propose.

Ex. It would be great if...

### Describe alternatives you've considered

I don't believe there is anyway to de engize a servo once a print begins. So the motor and electronics must be powered the full print time.

### Provide any additional context or information.

Ex. Photos, mockups, etc.

dc42, 2022-02-13:
The usual reason for servos wearing out quickly is that you have commanded
them beyond their working travel. Some servos are not very accurate, so
it's best to avoid the extremes of their published range.

I think that sending M280 S0 will already de-energise the servos.

T3P3, 2023-05-04:
@disneytoy please check to see if M280 S0 fulfils your requirements, if not then Please start a thread on the forum at https://forum.duet3d.com/ so that we can gauge how much interest there is in this feature.

closed 2023-05-04 by T3P3