Duet3D/RepRapFirmware #559: [Bug]: Stall detection work only with R0 & R1

rockman256, 2022-02-17:
### Duet Forum Discussion Thread

https://forum.duet3d.com/category/30/beta-firmware

### Which Duet products are you using?

- [X] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

RepRapFirmware for Duet 2 WiFi/Ethernet 3.4.0rc1

### Duet Web Control Version

Duet Web Control 3.4.0-rc1

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
=== Diagnostics ===
RepRapFirmware for Duet 2 WiFi/Ethernet version 3.4.0rc1 (2022-02-09 10:26:56) running on Duet WiFi 1.02 or later
Board ID: 08DGM-9T6BU-FG3S8-6J9DG-3S46Q-TVJVD
Used output buffers: 3 of 24 (24 max)
=== RTOS ===
Static ram: 23852
Dynamic ram: 74328 of which 232 recycled
Never used RAM 13596, free system stack 132 words
Tasks: NETWORK(ready,14.3%,243) HEAT(notifyWait,0.1%,332) Move(notifyWait,0.0%,305) MAIN(running,85.6%,410) IDLE(ready,0.0%,30), total 100.0%
Owned mutexes: WiFi(NETWORK)
=== Platform ===
Last reset 00:00:51 ago, cause: software
Last software reset at 2022-02-18 00:03, reason: User, GCodes spinning, available RAM 13668, slot 0
Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x0041f000 BFAR 0xe000ed38 SP 0x00000000 Task MAIN Freestk 0 n/a
Error status: 0x04
Step timer max interval 0
MCU temperature: min 42.7, current 43.0, max 43.3
Supply voltage: min 23.9, current 24.0, max 24.1, under voltage events: 0, over voltage events: 0, power good: yes
Heap OK, handles allocated/used 0/0, heap memory allocated/used/recyclable 0/0/0, gc cycles 0
Events: 0 queued, 0 completed
Driver 0: pos 0, standstill, SG min 0
Driver 1: pos 0, standstill, SG min 0
Driver 2: pos 48000, standstill, SG min 0
Driver 3: pos 0, standstill, SG min n/a
[... 143 lines cut ...]
M582 T1
G4 P10000
M955 P0 C"spi.cs4+spi.cs3"
```

### Please upload the content of any other releveant macro files.

```gcode
m915
Driver 0: stall threshold 1, filter off, steps/sec 380 (38.0 mm/sec), coolstep 0, action on stall: none
Driver 1: stall threshold 1, filter off, steps/sec 380 (38.0 mm/sec), coolstep 0, action on stall: none
Driver 2: stall threshold 1, filter off, steps/sec 200 (2.0 mm/sec), coolstep 0, action on stall: none
Driver 3: stall threshold 1, filter off, steps/sec 200 (7.7 mm/sec), coolstep 0, action on stall: none
Driver 4: stall threshold 1, filter off, steps/sec 200 (7.7 mm/sec), coolstep 0, action on stall: none
```

### Details specific to your printer.

DIY CoreXY

### What happened?

stall detection work only at homing. And when i use R1 - it can report me about stall detection. But, when i try set R2 or R3 - is nothing happend. it just keeps printing, don't pause the print. 
[image]
[image]

dc42, 2022-02-18:
This is a known issue that has already been resolved in the source code for forthcoming RC2 release. See https://github.com/Duet3D/RepRapFirmware/wiki/Changelog-RRF-3.x-RC#reprapfirmware-340rc2.

dc42, 2022-02-18:
PS - the fix is currently awaiting testing. If you wish to help with thus, you can find a preview 3.4.0rc1+1 build of RRF at https://www.dropbox.com/sh/2dt7sbqpx6l74np/AADn4-lpcil1iqnWKkiVri3Ia?dl=0.

dc42, 2022-03-29:
Fixed in release 3.4.0.

closed 2022-03-29 by dc42

kgolger, 2022-08-20:
A hello to the developers!

Same error on Release 3.4.1 on my Duet2 Wifi.
If M915 was invoked with parameter R2 or R3 then no action was taken on a driver stall as if R0 had been used. R1 is working.
rehome.g and driver_stall.g are stored in the /sys/ folder. Stallguard parameter are calibrated (M915 X Y S9 H200 F0 R3). Homing is without stallguard. M915 is active during printing (started from SD-card). Printing speed is 100-200mm/s. Stepper are 1.8deg (200steps/rot) with 160 esteps.

M915 messages:
<html><body>

20.8.2022, 14:57:07 | 
Driver 0: stall threshold 9, filter off, steps/sec 200 (39.9 mm/sec), coolstep 0, action on stall: run macro
Driver 1: stall threshold 9, filter off, steps/sec 200 (39.9 mm/sec), coolstep 0, action on stall: run macro
Driver 2: stall threshold 1, filter off, steps/sec 200 (8.0 mm/sec), coolstep 0, action on stall: none
Driver 3: stall threshold 1, filter off, steps/sec 200 (8.1 mm/sec), coolstep 0, action on stall: none
Driver 4: stall threshold 1, filter off, steps/sec 200 (40.0 mm/sec), coolstep 0, action on stall: none
-- | --

</body>
</html>

After these messages - no macro executing:
<html><body>

20.8.2022, 14:58:23 | Warning: Driver 0 stall
-- | --
20.8.2022, 14:58:15 | Warning: Driver 0 stall Warning: Driver 0 stall Warning: Driver 0 stall
20.8.2022, 14:57:44 | Warning: Driver 1 stall
20.8.2022, 14:57:40 | Warning: Driver 1 stall Warning: Driver 1 stall
20.8.2022, 14:57:35 | Warning: Driver 1 stall Warning: Driver 1 stall

</body>
</html>

Did I miss something?

Thanks for help!

wbr,
Karl

T3P3, 2022-08-20:
Please use our forum for support (forum.duet3d.com).

Check the release notes about the change to events, also documentation here:
https://docs.duet3d.com/User_manual/RepRapFirmware/Events