Duet3D/RepRapFirmware #565: [Bug]: G30 does not respect K1 when K0 triggers

greenaar, 2022-03-31:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/27983/g30-k1incorrectly-triggers-with-k0-rrf-3-4

### Which Duet products are you using?

- [X] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.4.0

### Duet Web Control Version

3.4.0

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
=== Diagnostics ===
RepRapFirmware for Duet 2 WiFi/Ethernet version 3.4.0 (2022-03-15 18:58:31) running on Duet WiFi 1.02 or later
Board ID: 0JD0M-9X6JA-JUNSN-6JKDL-3SN6S-K9NRS
Used output buffers: 7 of 24 (24 max)
=== RTOS ===
Static ram: 23868
Dynamic ram: 74992 of which 0 recycled
Never used RAM 10004, free system stack 106 words
Tasks: NETWORK(ready,13.3%,240) HEAT(notifyWait,0.0%,313) Move(notifyWait,1.9%,283) MAIN(running,84.5%,440) IDLE(ready,0.2%,30), total 100.0%
Owned mutexes: WiFi(NETWORK)
=== Platform ===
Last reset 00:46:59 ago, cause: power up
Last software reset at 2022-02-24 06:34, reason: User, GCodes spinning, available RAM 10772, slot 2
Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x0041f000 BFAR 0xe000ed38 SP 0x00000000 Task MAIN Freestk 0 n/a
Error status: 0x04
Aux0 errors 0,0,0
Step timer max interval 0
MCU temperature: min 13.6, current 32.8, max 33.4
Supply voltage: min 23.8, current 24.0, max 24.6, under voltage events: 0, over voltage events: 0, power good: yes
Heap OK, handles allocated/used 99/1, heap memory allocated/used/recyclable 2048/30/0, gc cycles 0
Events: 0 queued, 0 completed
Driver 0: ok, SG min 0
Driver 1: ok, SG min 0
Driver 2: standstill, SG min n/a
[... 201 lines cut ...]

CoreXY, modified tronxy X5SA pro. 24.  Duet 2 Wifi.

Switches on X and Y
Inductive probe on hotend (K0) (left side of nozzle)
Z offset switch attached to heatbed (K1) (right side of heatbed)

### What happened?

Expected result

When running homeall.g/homez.g (homez is just the bottom part of homeall), this should be probing the Z switch using K1.
The switch is just off the bed, so K0 will also trigger.

Sometimes when K0 triggers the homing action finishes, as if it has triggered K1, but it has not.

This tells me that G30 K1 also triggers on K0, or that I have somehow misread the documentation and missed something.

Observed result

G30 K1 should not accept K0 triggering as valid.

Steps to reproduce

Set up 2 probes, k0 and k1.  Force trigger K0, then try to home against K1.

AndyEveritt, 2023-06-06:
Could not reproduce in 3.5.0-b3

greenaar, 2023-06-08:
closing the issue, i've moved off of RRF to klipper and different hardware.

closed 2023-06-08 by greenaar