Duet3D/RepRapFirmware #576: [FeatureRequest]:

flobler, 2022-07-17:
### Is your feature request related to a problem? Please describe.

We are trying to create a very user-friendly way for Prusa MK3 Bear users to move from Einsy/Prusa Firmware to the Duet Mini/RRF. For this purpose we are trying to simplify setup and calibration of the printer via macros (e.g. Self Test or Calibration Wizard). Part of this is also to simplify first layer calibration as first layer problems are a very common problem leading to a lot of support requests.

I have created a macro that will allow users to adjust the first layer height by utilizing babystepping via a print. The print is a simple flat square in the middle of the bed. At the end of the print the adjustment will be automatically saved to config-override.g to make it persistent and independent of config.g updates that we will push to our users.

My problem is that we have to instruct users to open the babystepping dialog to make their adjustments. When the macro for first layer calibration is issued and the print started, a user needs to perform the following steps on the PanelDue:

1. Press "Status" on the PanelDue
2. Press "Baby step" on the PanelDue
3. Only then the user is able to make adjustments via the baby step dialogue

This is something many users will not know when they get started.

### Describe the solution you propose.

It would be great if there was a gcode command that will open the Baby step adjustment dialogue.

In this scenario the window would open automatically and all the user would be required to do is to make necessary adjustments via the buttons on the dialogue.

This is much more intuitive and will simplify this whole process.

### Describe alternatives you've considered

So far the workaround would be to explain to users what they need to do to open the Baby step window before they issue the macro. 

Problem with this is that many users will not read the documentation properly, then will issue the macro and are then confused how/where to make their adjustments.

### Provide any additional context or information.

Ex. Photos, mockups, etc.

T3P3, 2022-07-17:
I think this would need the PanelDue firmware updated to accept the "open dialog" command.

dc42, 2023-05-05:
I believe there is already a way to achieve this:

- Read the Z height from the object model and save it in a variable
- Execute a M291 command to open a blocking dialog with Z axis jog enabled
- After the M291 completes (because the user has closed the dialog, read the Z height again and execute a M292 command to increase the babystepping amount by the change in Z height

Please try this and report back.

closed 2025-02-23 by T3P3