Duet3D/RepRapFirmware #600: [FeatureRequest]: WPA3 support - Protected management frames (PMF)

eMPai-git, 2023-02-23:
### Is your feature request related to a problem? Please describe.

We are running a 3D printer with a Duet 2 WiFi board. Our wireless network environment requires to use Protected management frames (PMF) IEEE 802.11w-2009 since there are security concerns not using it. Because of this, it is not possible to connect the Duet 2 WiFi to our wireless network.

### Describe the solution you propose.

The ESP8266 RTOS Software Development Kit supports WPA3 and PMF since April 8, 2021. See the [release notes](https://github.com/espressif/ESP8266_RTOS_SDK/releases/tag/v3.4).
My preferred solution would be added WPA3 support.

### Describe alternatives you've considered

We discussed the possibility to set 802.11w to optional. However, this was ultimately rejected.

### Provide any additional context or information.

[ESP8266_RTOS_SDK Release v3.4](https://github.com/espressif/ESP8266_RTOS_SDK/releases/tag/v3.4)

dc42, 2023-02-24:
Have you tried the version 2.1beta3 DuetWiFiFirmware that we release with RRF 3.5beta2? That version uses a recent version of the ESP8266 SDK.

dc42, 2023-05-04:
@eMPai-git WPA3 is disabled in the ESP8266 build due to limitations of the Expressif SDK and/or insufficient RAM on the ESP8266. It may be working on the ESP32S3 build which is used by the optional WiFi module for the Duet MB6HC version 1.02.

T3P3, 2025-02-03:
We will not be supporting this on ESP8266 based Wifi modules due to  limitations of the Expressif SDK and/or insufficient RAM

closed 2025-02-03 by T3P3