Duet3D/RepRapFirmware #602: [FeatureRequest]: array unpack operator

douarvil-yamaichi, 2023-02-27:
### Is your feature request related to a problem? Please describe.

Here is a concrete example:
I have a variable number of tools which physical drive number are saved as `set global.ioMotEx = {0.1, 1.1, 2.0,}`.

Extruder drive mapping without unpack operator:
`M584 E{global.ioMotE[0], global.ioMotE[1], global.ioMotE[2]}`

Without unpack operator code of the M584 mapping is needlessly long and repetitive. M584 content must be manually adjusted whenever the length of `global.ioMotEx` changes. This is a breeding ground for preventable bugs and subsequent headaches.

### Describe the solution you propose.

Extruder drive mapping with an hypothetical "~" unpack operator:
`M584 E{~global.ioMotE}`

With the unpack operator the code is shorten, the risk for preventable bug is reduced, the code readability and modularity is greatly enhanced.

dc42, 2023-03-03:
It's more likely that we will support just `M584 E{global.ioMotE}`. I see no need for a new operator.

douarvil-yamaichi, 2023-03-09:
Thank you. Got it, thank you.
Should I "Close as not planned"?