Duet3D/RepRapFirmware #812: file not found "MessageFormats.h" due to WiFi/Wifi spelling .cproject

tobbelobb, 2023-04-28:
I got a file not found "MessageFormats.h" error when trying to build today.

Turned out the DuetWifiSocketServer was spelled with a capital F (WiFi) in the .cproject file. Changing it to lowercase f (Wifi) fixed the issue for me.

https://github.com/Duet3D/RepRapFirmware/blob/0055cbdf32983bad645181ab03889740ee0edd97/.cproject#L612

The DuetWifiSocketServer repo itself uses lowercase, so we should make the two match.

dc42, 2023-04-29:
The repository name is DuetWiFiSocketServer (with uppercase F), so that should be the name of the folder containing that project in your Eclipse workspace, as it is mine. I suspect that you created the folder with the wrong name.

closed 2023-04-29 by dc42

tobbelobb, 2023-04-30:
You're right. This taught me something about Github. In my bash history:
```
$ git clone git@github.com:Duet3d/DuetWifiSocketServer.git
```
In the repo itself:
```
$ git remote -v
origin	git@github.com:Duet3d/DuetWifiSocketServer.git (fetch)
origin	git@github.com:Duet3d/DuetWifiSocketServer.git (push)
```

... both lowercase f. But going to the website, one can see that Github silently fixes my typo and prints the fixed version back in the tab title. 
[image]

So it silently does the same fix to my git requests from the command line, and there's no warning or anything.