Duet3D/RepRapFirmware #817: [FeatureRequest]:

T3P3, 2023-05-03:
### Is your feature request related to a problem? Please describe.

Allow monitoring of closed loop motors attached to the 1HCL and M23CL.

If the information was available in the Object Model then plugins and DWC could use it.

### Describe the solution you propose.

Report the closed loop current error, phase shift and potentially coil A and B current in the OM for each closed loop motor. probably at a similar frequency to temperature and other continuously updated data.

### Describe alternatives you've considered

A third party current monitor could potentially be used on the 1HCL, on the Motor23CL this would not be easy to use.

dc42, 2023-05-04:
I don't see how this would be useful as the request now stands, because these quantities vary very rapidly, so taking a snapshot every 250ms wouldn't give a true indication of what is happening.

T3P3, 2023-05-05:
potentially a running mean, and a peak reported every 250ms could be useful?

T3P3, 2024-02-15:
implemented in 3.5

closed 2024-02-15 by T3P3