Duet3D/RepRapFirmware #822: [FeatureRequest]: Filament sensor/monitor used for load/unload

T3P3, 2023-05-03:
### Is your feature request related to a problem? Please describe.

It would be useful if the filament monitor or a filament sensor could be configured to act as a filament "endstop"/trigger (present/not present) to allow for it to be used in filament load and unload macros.

### Describe the solution you propose.

Multiple potential ways to do this:
Provide a filament present/not present state in the OM
Provide a trigger based on filament movement/no movement or present/not present

### Describe alternatives you've considered

For a switch based presence sensor it can be reconfigured as a trigger during the load/unload macro and then configured back to a filament sensor at the end of the macro.

gnydick, 2023-05-19:
I do this currently in macros. What is the enhancement being requested?

dc42, 2023-05-19:
Stall detection can also be used to load filament until it reaches the hot end.

Using a filament detection switch to control unloading filament would be appropriate if the filament sensor was located some distance before the extruder drive. I don't see how a filament sensor could be used to control loading filament.

gnydick, 2023-05-19:
Look at the orbiter projects filament sensor

dc42, 2023-05-21:
@gnydick please provide a link.

gnydick, 2023-05-21:
https://www.orbiterprojects.com/orbiter-filament-sensor/

gnydick, 2025-08-30:
> The aim here is to push the filament into the tube leading to the
> extruder, when it reaches the filament monitor (just before the extruder)
> the user can then push slightly more and the load macro can take over and
> complete the load (potentially using stall detection). Currently one has to
> position the filament though the filament monitor, ensure it is engaging
> the extruder teeth and then start a filament load macro.

I understand, but literally all of the filament loaders use a switch and
it's 100% reliable. I think you're overthinking it.

chrishamm, 2026-08-07:
Tested and working.