Duet3D/RepRapFirmware #835: Closed Loop: jolt when switching mode

T3P3, 2023-05-03:
When switching into closed lop mode, read the microstep position and set the motor currents to be the same phase, to avoid a jolt

Now there is no jolt at the start, but there is a jolt at the end, at least when the encoder is working in reverse polarity

dc42, 2023-05-24:
There is a more severe jolt when switching from closed to open loop mode after doing some movement in closed loop mode.

dc42, 2023-06-12:
Fixed this apart from the jolt when switching form closed loop back to open loop after doing movement. To avoid the jolt we could:
- Disable the driver
- Switch the driver out of XDIRECT mode
- Send step pulses to the driver until the microstep counter is at the correct phase position
- Re-enable the driver
Alternatively, we could use XDIRECT mode even for open loop operation and never generate step/dir signals, then we would never need to sync the microstep counter with the phase position.

T3P3, 2026-08-24:
could be done as part of #839