Duet3D/RepRapFirmware #838: Closed Loop: Improve CAN-FD data transmission efficiency

T3P3, 2023-05-03:
When collecting data, pack both motor currents into a single 32-bit word, and unpack them at the other end. This will make room for another variable to be recorded. Likewise we could combine desired and measure step phase into a single 32-bit word.

On the main board, when writing data to file, write only the number of decimal places needed for that type of data

Currently, reduced the number of decimal places from 6 to 2 to reduce data size and speed up writing to file, but more reduction in data size is needed

T3P3, 2023-05-10:
Also tied to supporting a Encoder DB on the 1HCL

dc42, 2023-05-23:
The data is now sent using smaller types where possible, and the V and A terms can be selected for inclusion.

closed 2023-05-23 by dc42