Duet3D/RepRapFirmware #840: Can't enable both Ethernet and WiFi in config.g on MB6HC 1.02 witt WiFi module

dc42, 2023-05-07:
### Duet Forum Discussion Thread

n/a

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [X] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

RRF 3.5beta3+

### Duet Web Control Version

3.5

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
n/a
```

### Please upload the content of your config.g file.

```gcode
n/a
```

### Please upload the content of any other releveant macro files.

```gcode
n/a
```

### Details specific to your printer.

n/a

### What happened?

The following lines in config.g don't enable the WiFi interface:

M552 I0 S1                          ; enable Ethernet
M552 I1 S1                          ; enable WiFi (doesn't seem to work)

Sending M552 I1 S1 via USB from YAT does enable the WiFi interface, although one "Authentication failed" response is returned (known issue with version 2.1 WiFi server).

dc42, 2023-05-19:
Fixed in 3.5.0-beta.4 source code

dc42, 2023-12-13:
@Lino77 please try installing wifi firmware 2.1beta 6 from https://github.com/Duet3D/WiFiSocketServerRTOS/releases/tag/2.1beta6.

Lino77, 2023-12-13:
Thanks, it now works, after renaming it to DuetWiFiModule_32S3.bin I was able to flash it, DuetWiFiServer_32S3.bin is probably incorrectly labeled on the github page. The upload via DWS renames the file DuetWiFiModule_32S3.bin to DuetWiFiServer.bin.

Lino77, 2023-12-13:
Is it normal that the wifi module starts almost 2 minutes later

2023-12-13 22:11:52 [warn] Date and time set at power up + 00:00:04
2023-12-13 22:11:52 [warn] HTTP client 192.168.178.40 login succeeded (session key 0)
2023-12-13 22:13:20 [warn] WiFi module started
2023-12-13 22:13:20 [warn] WiFi module started
2023-12-13 22:13:27 [warn] WiFi module is connected to access point lunipu24, IP address 192.168.178.86

Lino77, 2024-03-30:
The problem now exists again in 3.5.0-rc.3  wifi firmware 2.1beta 7 ,
via usb it start after sending M587 

M587
Remembered networks: none
ok
WiFi module started
WiFi module is connected to access point lunipu24, IP address 192.168.178.86

Lino77, 2024-04-08:
@dc42 okay but the WiFi module does not start by itself, only after I enter M587 via USB

rechrtb, 2024-04-08:
Hi @Lino77  just to clarify, this is on an MB6HC + Duet3 WiFi module?

Lino77, 2024-04-09:
@rechrtb  

Yes

rechrtb, 2024-04-15:
Hi @Lino77, since there is now a new release candidate, are you able to upgrade to that and see if the problem still persists?

Lino77, 2024-04-15:
I will do it and report

Lino77, 2024-04-17:
okay RC 5 works perfectly, the best in terms of WLAN

rechrtb, 2024-04-17:
Hi @Lino77, since it seems to be working for you now, will close this issue again.

closed 2024-04-17 by rechrtb