Duet3D/RepRapFirmware #844: Duet3 Mini: allocate EXINT6 and EXINT9 to spi.cs1 and spi.cs2 so that an accelerometer can be connected to temp daughterboard connector

dc42, 2023-05-09:
### Is your feature request related to a problem? Please describe.

EXINTs 6 and 9 are currently allocated to VBUS PB06) and to the encoder push button (PB09) but we don't currently use either of them. If instead we allocate them to spi.cs1 (PD11) and spi.cs2 (PC07) then those pins could be used as inputs. Using spi.cs1 for the accelerometer INT output would facilitate the use of a common SPI accelerometer cable with Duet 3 and Duet 3 6HC/6XD.

### Describe the solution you propose.

As above. Need to check that PC07 really does use EXINT9 in case it is a typo on the datasheet (it would be more obvious if it used EXINT7).

dc42, 2023-05-09:
The EXINTs have been reassigned in RRF 3.5beta4 source. Still need to test EXINT9 on spi.cs2.

T3P3, 2023-05-26:
Tested working well with 1m cable on mini 5+

closed 2023-05-26 by T3P3