Duet3D/RepRapFirmware #847: G1 H1 E moves not working in 3.5.-beta.3

dc42, 2023-05-10:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/32153/m915-feature-is-loss-after-3-5-0-beta2-version-on-extruder

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

qq

### Duet Web Control Version

qq

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [ ] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122 Report
```

### Please upload the content of your config.g file.

```gcode
Config.g
```

### What happened?

qq

dc42, 2023-05-10:
See https://github.com/Duet3D/RepRapFirmware/pull/843 for a user-suggested fix.

alexjx, 2023-06-17:
@dc42 David, you might be influenced by my patch. It seems that following code is incorrect. We should allow checkEnstop moves instead of skip it.

My testing of the latest 3.5.0 beta4 shows it's not working as intended

```cpp
	// Check for cold extrusion/retraction. Do this now because we can't read temperatures from within the step ISR, also this works for CAN-connected extruders.
	// Don't check if it is a special move (indicated by flags.checkEndstops) because the 'tool' variable isn't valid for those moves
	if (!flags.checkEndstops || Tool::ExtruderMovementAllowed(tool, directionVector[drive] > 0, extruder))
	{
```

which should be 

```cpp
	// Check for cold extrusion/retraction. Do this now because we can't read temperatures from within the step ISR, also this works for CAN-connected extruders.
	// Don't check if it is a special move (indicated by flags.checkEndstops) because the 'tool' variable isn't valid for those moves
	if (flags.checkEndstops || Tool::ExtruderMovementAllowed(tool, directionVector[drive] > 0, extruder))
	{
```

dc42, 2023-06-18:
Thanks @alexjx, I will fix this in the next commit.

T3P3, 2023-08-02:
@alexjx please can you test to se eif this is now working for you in RRF 3.5.0-beta4

alexjx, 2023-08-07:
@T3P3 I'm sorry, I dont have access to my printers in this two weeks. Can it wait till end of this weekend?

T3P3, 2023-08-07:
Plase test and let us know as soon as you can, we may release the next beta/rc1 in between.

alexjx, 2023-08-07:
OK, I will give it a try by this Sunday.

alexjx, 2023-08-14:
@T3P3 I could confirm this works in latest 3.5-dev branch.

T3P3, 2023-08-14:
Thanks, will close this as fixed

closed 2023-08-14 by T3P3