Duet3D/RepRapFirmware #850: [Bug]: Issue with forgetting all Wi-Fi networks (M588) in beta version

JPluess, 2023-05-12:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/32318/safety-timer?_=1683870830019

### Which Duet products are you using?

- [X] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.5 (3)

### Duet Web Control Version

dwc 3.5

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122 Report
=== Diagnostics ===
RepRapFirmware for Duet 2 WiFi/Ethernet version 3.5.0-beta.3 (2023-04-14 11:27:37) running on Duet WiFi 1.02 or later
Board ID: 0JD0M-9K662-MG5TD-6JKDL-3S06T-K8JUZ
Used output buffers: 1 of 26 (2 max)
Error in macro line 48 while starting up: in file macro line 48: M591: missing parameter 'D'
=== RTOS ===
Static ram: 22192
Dynamic ram: 76384 of which 12 recycled
Never used RAM 11700, free system stack 186 words
Tasks: NETWORK(ready,14.0%,373) HEAT(nWait,0.0%,330) Move(nWait,0.0%,364) MAIN(running,85.1%,234) IDLE(ready,0.8%,29), total 100.0%
Owned mutexes: USB(MAIN)
=== Platform ===
Last reset 00:24:15 ago, cause: power up
Last software reset at 2021-01-05 19:01, reason: User, Gcodes spinning, available RAM 14840, slot 0
Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x00400000 BFAR 0xe000ed38 SP 0x00000000 Task MAIN Freestk 0 n/a
Error status: 0x00
Aux0 errors 0,0,0
MCU temperature: min 30.3, current 32.0, max 32.5
Supply voltage: min 1.4, current 1.7, max 1.8, under voltage events: 0, over voltage events: 0, power good: no
Heap OK, handles allocated/used 99/2, heap memory allocated/used/recyclable 2048/64/0, gc cycles 0
Events: 1 queued, 1 completed
Driver 0: ok, SG min n/a
Driver 1: ok, SG min n/a
Driver 2: ok, SG min n/a
Driver 3: ok, SG min n/a
Driver 4: ok, SG min n/a
Driver 5: 
Driver 6: 
Driver 7: 
Driver 8: 
Driver 9: 
Driver 10: 
Driver 11: 
Date/time: 1970-01-01 00:00:00
Cache data hit count 4294967295
Slowest loop: 3692.05ms; fastest: 0.15ms
I2C nak errors 0, send timeouts 0, receive timeouts 0, finishTimeouts 0, resets 0
=== Storage ===
Free file entries: 10
SD card 0 detected, interface speed: 20.0MBytes/sec
SD card longest read time 2.6ms, write time 0.0ms, max retries 0
=== Move ===
DMs created 83, segments created 0, maxWait 0ms, bed compensation in use: none, comp offset 0.000
no step interrupt scheduled
=== DDARing 0 ===
Scheduled moves 0, completed 0, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
=== Heat ===
Bed heaters 0 -1 -1 -1, chamber heaters -1 -1 -1 -1, ordering errs 0
=== GCodes ===
Movement locks held by null
HTTP is idle in state(s) 0
Telnet is idle in state(s) 0
File is idle in state(s) 0
USB is ready with "M122" in state(s) 0
Aux is idle in state(s) 0
Trigger is idle in state(s) 0
Queue is idle in state(s) 0
LCD is idle in state(s) 0
Daemon is idle in state(s) 0
Autopause is idle in state(s) 0
Q0 segments left 0
Code queue 0 is empty
=== Network ===
Slowest loop: 3692.12ms; fastest: 0.00ms
Responder states: HTTP(0) HTTP(0) HTTP(0) HTTP(0) FTP(0) Telnet(0)
HTTP sessions: 0 of 8
= WiFi =
Network state is active
Module is connected to access point 
Failed messages: pending 0, notready 0, noresp 0
Firmware version 2.1beta4
MAC address ***
Module reset reason: Turned on by main processor, Vcc 3.43, flash size 2097152, free heap 46528
WiFi IP address ***
Signal strength -63dBm, channel 1, mode 802.11n, reconnections 0
Clock register 00002002
Socket states: 0 0 0 0 0 0 0 0
```

### What happened?

Title: Issue with "Forget All Wi-Fi Networks" (M588) command in beta version

Description:
I am experiencing an issue when using the "Forget All Wi-Fi Networks" command (M588 S"*") in the beta version of the firmware on my DuetWifi board. After sending the command, it returns "ok," but the Wi-Fi module still appears to be connected.

Expected Behavior:
I expect that after using the M588 command to forget all Wi-Fi networks, the M552 command should indicate that no networks are stored. However, the M552 command still shows the Wi-Fi module as connected to an access point, even though the router does not list it as connected.

Steps to Reproduce:

1.     Connect to the Duet3D board via USB.
2.     Send the command "M588 S"*"" to forget all Wi-Fi networks.
3.     Send the command "M552" to check the network status.

Actual Results:
After step 3, the M552 command returns:
"WiFi module is connected to access point **, IP address 192.*"
Additionally, the LED on the Wi-Fi module remains solid blue.

Expected Results:
I expect the M552 command to return "No networks stored", indicating that no Wi-Fi networks are stored. Furthermore, the LED on the Wi-Fi module should be turned off.

Additional Information:
I have also tested connecting with incorrect Wi-Fi passwords, and the Wi-Fi module still claims to be connected. This suggests that there might be an issue with the process of forgetting all Wi-Fi networks.

[image]

T3P3, 2023-05-24:
from the report it appears that this issue occurs only if the network is currently connected when the M588 command is sent. 

This behaviour may be present in 3.4.5 as well.

A work around may be to always send M552 S-1 after M588 - please check if this works.

AndyEveritt, 2023-05-28:
Able to replicate in RRF3.4.5

- The device does appear to forget the network
  - Confirmed by calling `M587` after the `M587 S"*"`
- The device does remain connected to the network with DWC accessable until either:
  - Duet/RRF is restarted
  - WiFi is disabled with `M552 S0`/`M552 S-1`

I believe this is intended behaviour.

[image]

@T3P3 your solution (calling `M552 S-1` first) would not work as the WiFi module needs to be on when calling `M588`.
> The M588 command will fail if the WiFi module has not yet been taken out of reset. So if the WiFi module has not been started, send M552 S0 to put it in idle mode first. M588 does not work from within config.g at startup. (https://docs.duet3d.com/User_manual/Reference/Gcodes#m588-forget-wifi-host-network)

[image]

AndyEveritt, 2023-05-28:
Behaviour is the same in RRF3.5.0-b3

dc42, 2023-06-07:
Updated the documentation.

closed 2023-06-07 by dc42

AndyEveritt, 2023-06-11:
Docs updated