Duet3D/RepRapFirmware #854: New PWM code in RRF 3.5 doesn't allow PWM frequencies as low as 3.4 did

dc42, 2023-05-21:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/32487/3-5-0-beta-3-minimum-pwm-frequency

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.5.0-beta.3

### Duet Web Control Version

3.5.0-beta.3

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [ ] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122 Report
```

### Please upload the content of your config.g file.

```gcode
Config.g
```

### What happened?

The minimum PWM frequency on the 6HC used to be about 1.6Hz on outputs that use the PWM frequency. in 3.5.0-beta3 it is about 3Hz.

dc42, 2023-05-21:
Implement a probably fix in CoreN2G.

T3P3, 2023-05-26:
tested on mini 5+ with latest build. Using:
M950 P0 C"out5" Q1
then
M42 P0 S0.5
resulted in approx 1Hz flashing of LED connected to out5.

dc42, 2023-05-27:
Unfortunately that's not a valid test, because the code change only affects boards using the SAM4E, SAME7x or SAM4S, and it only affects those output ports that use the PWM peripheral. For example, on the 6HC it affects laser/vfd, out5, out2, out8, out1, out0.

T3P3, 2023-06-09:
OP on the forum has confirmed fixed

closed 2023-06-09 by T3P3

Pourya-3deo, 2024-04-18:
I have the same problem, I need to have 10 hz frequency but the lowest I could get is 17.88 Hz, I measured it with an oscilloscope.

dc42, 2024-04-19:
@Pourya-3deo that is a different issue, it's a limitation of the PWM-capable ports that are drive by a timer/counter.