Duet3D/RepRapFirmware #858: Action by CAN-FD expansion board on disconnection of CAN-FD bus

T3P3, 2023-05-31:
Are you planning on doing anything on the expansion board side if communication seems to be lost with the main board?

_Originally posted by @gloomyandy in https://github.com/Duet3D/RepRapFirmware/issues/769#issuecomment-1561905065_

T3P3, 2026-09-16:
needs to be handled in main board in expansion mode.

AndyEveritt, 2026-09-16:
[image]

T3P3, 2026-09-16:
Forum threads:
https://forum.duet3d.com/topic/32348
https://forum.duet3d.com/topic/31988

chrishamm, 2026-09-19:
The expansion side of this does work: `setConnectionTimeout` (6071) is defined in CANlib and handled in Duet3Expansion since 3.7.0-beta.2. What was missing is a main board running in expansion mode, which had no case for the message and no sync-loss heater shutdown either.

Fixed on 3.7-ch in 1615410dd. A 6HC or Mini in expansion mode now accepts `M959 T`, switches its heaters off after the timeout when time sync is lost, re-announces itself with the reconnect flags (those were uninitialised before, so they could cause spurious `expansion-reconnect` events), and reports the timeout and the counters in `M122`. `M959 B<n> T<n>` also no longer updates the main board's own copy of the timeout when the request fails, which is why the object model showed 30 in that run.