Duet3D/RepRapFirmware #859: [Bug]: Printer hang when printing

AndyEveritt, 2023-05-31:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/32084/how-to-get-dwc-to-display-what-line-of-gcode-it-is-on/41?_=1685560419092

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [X] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

Reprapfirmware 3.5.0-b3

### Duet Web Control Version

3.5.0-b3

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [X] Yes I use a SBC.
- [ ] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
Duet hangs so can not get report
```

### Please upload the content of your config.g file.

```gcode
Config.g
```

### Details specific to your printer.

E3D Toolchanger

### Links to additional info.

https://github.com/Duet3D/RepRapFirmware/issues/620
https://forum.duet3d.com/topic/32084/how-to-get-dwc-to-display-what-line-of-gcode-it-is-on/41?_=1685560419092

### What happened?

Expected result
- Print finishes successfully

Observed result
- Duet hangs after homing Z

Steps to reproduce
- create a print file that contains the following gcode
```
M140 S85 ; set bed temperature
M104 S240 ; set temperature
;TYPE:Custom
; custom gcode: start_gcode
T-1
M557 X80.4905:219.51 Y63.0512:143.076 S20
G28 Z
G29
G29 S1
G10 P0 S240 R240
T0
M116 ; wait for temperatures to reach target
G1 Z5 F3000 ; lift
G1 X300 Y10 F10000 ; move to prime
G1 Z0.2 F3000 ; get ready to prime
G92 E0 ; reset extrusion distance
G1 Y80 E20 F600 ; prime nozzle
G1 Y100 F5000 ; quick wipe
G1 Z0 ; prevent oozing
; custom gcode end: start_gcode
M107
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
; custom gcode: start_filament_gcode
; Filament gcode
; custom gcode end: start_filament_gcode
;_TOOLCHANGE 0
M109 S240 ; set temperature and wait for it to be reached
M190 S85 ; set bed temperature and wait for it to be reached
```
- Run print file

Removing any aspect of the file prevents the issue from occuring

AndyEveritt, 2023-05-31:
[config.zip](https://github.com/Duet3D/RepRapFirmware/files/11618207/config.zip)

AndyEveritt, 2023-05-31:
This is not an issue in RRF3.4.5

chrishamm, 2023-05-31:
Thanks for logging this, I'll try to reproduce it. Since you are using an SBC and just `M122` doesn't produce a result, maybe `M122 "DSF"` does. That's a special M122 command for situations like this.

dc42, 2023-05-31:
Is this issue reproducible in standalone mode too?

AndyEveritt, 2023-06-01:
I will need to verify it on standalone. This will require some hardware to be added to the board though. I think 2 microswitches and a hotend should be enough

chrishamm, 2023-06-01:
I am quite sure this is already fixed upstream. Nevertheless your demo file let me reproduce another bug which could affect RRF in standalone mode, too.