Duet3D/RepRapFirmware #861: Duet 3 Mini WiFi crash in network interface code

dc42, 2023-06-05:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/32537/extruder-stops-extruding-again-reboots-randomly-rrf-3-5b3

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [X] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.5beta3+

### Duet Web Control Version

3.5beta3

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122 Report
```

### Please upload the content of your config.g file.

```gcode
Config.g
```

### What happened?

Duet is resetting randomly within the WiFi code.

dc42, 2023-06-05:
Traced it to writing data via a null buffer pointer in WiFiSocket::ReceiveData. However, RRF already guards that code with a check for a null pointer. Temporarily added checks on the value of the register concerned around the calls made by that function.

T3P3, 2024-05-17:
updated thread here:
https://forum.duet3d.com/topic/33998/reboots-crashes-rrf-3-5-0-rc1/135

closed 2025-08-25 by T3P3