Duet3D/RepRapFirmware #864: In SBC mode the network IP address is not available to the 12864 display system

dc42, 2023-06-08:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/32638/12864-lcd-n534-is-not-work-in-duet3-mini5-ver-3-4-5

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.4.5

### Duet Web Control Version

3.4.5

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [X] Yes I use a SBC.
- [ ] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122 Report
```

### Please upload the content of your config.g file.

```gcode
Config.g
```

### What happened?

The following 12864 menu entry does not work in SBC mode:

```
text R54 C17 F0 T"IP:"
value N534 W30
```

chrishamm, 2023-06-08:
The same limitation applies to PanelDue.

T3P3, 2024-05-22:
@droftarts please document that the IP address is not available on 12864 displays in SBC mode

dc42, 2024-06-30:
Now that DSF provides the IP address to the Duet as of firmware 3.5.2 it should be straightforward to fix this. It should already work on PanelDue.

chrishamm, 2024-11-21:
Second fix is 68d796e6. I haven't tested this code yet.

chrishamm, 2025-04-23:
Confirmed fixed.

closed 2025-04-23 by chrishamm