Duet3D/RepRapFirmware #868: [Bug]: Too Many Message Boxes causes erroneous errors and in some cases a stack overflow

jaysuk, 2023-06-13:
### Duet Forum Discussion Thread

see slack

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [X] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.5b4

### Duet Web Control Version

3.5b4

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
=== Diagnostics ===
RepRapFirmware for Duet 3 Mini 5+ version 3.4.5 (2022-11-30 19:41:16) running on Duet 3 Mini5plus WiFi (standalone mode)
Board ID: BTKMS-0N6KL-K65J0-409NU-NH02Z-H6F90
Used output buffers: 1 of 40 (27 max)
=== RTOS ===
Static ram: 103652
Dynamic ram: 112508 of which 0 recycled
Never used RAM 25552, free system stack 204 words
Tasks: NETWORK(notifyWait,14.6%,241) HEAT(notifyWait,0.0%,358) Move(notifyWait,0.0%,363) CanReceiv(notifyWait,0.0%,942) CanSender(notifyWait,0.0%,336) CanClock(delaying,0.0%,341) TMC(notifyWait,0.7%,115) MAIN(running,83.8%,425) IDLE(ready,0.0%,30) AIN(delaying,0.9%,272), total 100.0%
Owned mutexes:
=== Platform ===
Last reset 00:03:26 ago, cause: power up
Last software reset at 2023-06-11 20:13, reason: User, GCodes spinning, available RAM 12004, slot 1
Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x00000000 BFAR 0xe000ed38 SP 0x00000000 Task MAIN Freestk 0 n/a
Error status: 0x00
Aux0 errors 0,0,0
MCU revision 3, ADC conversions started 206451, completed 206450, timed out 0, errs 0
Step timer max interval 1488
MCU temperature: min 38.1, current 38.6, max 38.6
Supply voltage: min 24.0, current 24.1, max 24.1, under voltage events: 0, over voltage events: 0, power good: yes
Heap OK, handles allocated/used 0/0, heap memory allocated/used/recyclable 0/0/0, gc cycles 0
Events: 0 queued, 0 completed
Driver 0: standstill, SG min 0, read errors 0, write errors 0, ifcnt 11, reads 1041, writes 0, timeouts 0, DMA errors 0, CC errors 0
Driver 1: standstill, SG min 0, read errors 0, write errors 0, ifcnt 11, reads 1041, writes 0, timeouts 0, DMA errors 0, CC errors 0
[... 189 lines cut ...]
Error: M292: no active message box
Error: M292: no active message box
Error: M292: no active message box
Error: M292: no active message box
13/06/2023, 07:20:32	M292 S6
Error: M292: no active message box
Error: M292: no active message box
Error: M292: no active message box
Error: M292: no active message box
Error: M292: no active message box
13/06/2023, 07:20:32	M292 S4
Error: M292: no active message box
Error: M292: no active message box
Error: M292: no active message box
Error: M292: no active message box
Error: M292: no active message box
13/06/2023, 07:20:32	M292 S3
Error: M292: no active message box
Error: M292: no active message box
Error: M292: no active message box
Error: M292: no active message box
Error: M292: no active message box`

Steps to reproduce
run macro

dc42, 2023-07-21:
Fixed the M292 error messages, see [issue ](https://github.com/Duet3D/RepRapFirmware/issues/730)

dc42, 2023-08-01:
If the daemon.g file is modified to set the timeout to 15 seconds, then if the messages it generates are not acknowledged promptly, we get the stack overflows even without the macro file running. The reason is that the status of an input channel when it is awaiting an acknowledgment is idle, which causes RRF to think that daemon.g is t running, so it restarts.

dc42, 2023-08-01:
Fixed by checking that the daemon input channel is not in a stacked state and doesn't have a file open before we delay and then open/start daemon.g.

closed 2023-08-01 by dc42