Duet3D/RepRapFirmware #888: Stop segmenting moves after Z bed compensation taper completes

T3P3, 2023-08-02:
to improve input shaping performance.

chrishamm, 2023-08-04:
Changed in 0dec97fb78fec77e8161fd8e92e36a88b50da336

AndyEveritt, 2023-08-31:
Some initial testing suggests that there is no improvement to input shaping because of this change. Will try some more test objects.

The expected issue is that IS is not being applied on the deceleration of a move that too long (relative to the bed mesh spacing)

Tested by printing 2 cubes on the bed with a large spacing between them (multiple times the bed mesh spacing), and enabling a taper for the mesh compensation.

Ringing was noticed on the edge immediately after the travel move (expected because IS won't be applied), but no change was seen before or after the taper height.

closed 2023-08-31 by chrishamm