Duet3D/RepRapFirmware #899: Print jobs sometimes fail to complete

dc42, 2023-08-25:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/33151/does-not-completely-finish-the-last-layer-of-the-print-3-5-0-b4

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [X] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [X] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.5.0-beta.4

### Duet Web Control Version

3.5.0-beta.4

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
The following was taken on a toolchanger. Printing had stopped with the nozzle still on the print. DWC showed still printing but at 100%. Note that the File input is syncing but File2 is not. I have seen this same situation once before. After pressing Pause and Cancel, the tool crashed into the Z axis when the tool was being unloaded.

23/08/2023, 19:25:00 	m122
=== Diagnostics ===
RepRapFirmware for Duet 3 MB6HC version 3.5.0-beta.4+ (2023-08-15 14:30:07) running on Duet 3 MB6HC v1.01 (standalone mode)
Board ID: 08DJM-956BA-NA3TJ-6J1FG-3SN6P-TAAUU
Used output buffers: 3 of 40 (36 max)
=== RTOS ===
Static ram: 154852
Dynamic ram: 126156 of which 0 recycled
Never used RAM 61216, free system stack 117 words
Tasks: NETWORK(1,ready,27.3%,182) ETHERNET(5,nWait,0.1%,443) LASER(5,nWait,0.0%,174) HEAT(3,nWait,0.0%,323) Move(4,nWait,1.6%,154) CanReceiv(6,nWait,0.1%,670) CanSender(5,nWait,0.0%,328) CanClock(7,delaying,0.0%,349) TMC(4,nWait,8.9%,59) MAIN(1,running,61.9%,137) IDLE(0,ready,0.0%,30), total 100.0%
Owned mutexes: LwipCore(NETWORK)
=== Platform ===
Last reset 00:38:50 ago, cause: software
Last software reset at 2023-08-20 20:34, reason: MemoryProtectionFault mmarValid daccViol, Gcodes spinning, available RAM 64648, slot 2
Software reset code 0x0163 HFSR 0x00000000 CFSR 0x00000082 ICSR 0x0043c804 BFAR 0x00000041 SP 0x2041bc20 Task MAIN Freestk 2212 ok
Stack: 00000000 00000000 00000000 2042dfc8 00000000 00467455 004053be 41030000 00498519 2042dfb8 00000000 2042ecd8 00467455 2042e9f0 2041bcd4 2042ecd0 0046d76f fffc0001 2042e9ff 00000000 2042bb28 204309d8 204251bc 2042d328 0046dba9 20425bd8 2042e9f4
Error status: 0x00
Aux0 errors 0,0,0
MCU temperature: min 48.4, current 56.4, max 56.8
Supply voltage: min 24.1, current 24.2, max 24.2, under voltage events: 0, over voltage events: 0, power good: yes
12V rail voltage: min 12.0, current 12.1, max 12.2, under voltage events: 0
Heap OK, handles allocated/used 99/0, heap memory allocated/used/recyclable 2048/716/716, gc cycles 0
[... 131 lines cut ...]
Messages queued 75351, received 0, lost 0, boc 0
Longest wait 0ms for reply type 0, peak Tx sync delay 0, free buffers 18 (min 18), ts 41863/0/0
Tx timeouts 0,0,41862,0,0,33487 last cancelled message type 4514 dest 127
=== Network ===
Slowest loop: 15.28ms; fastest: 0.00ms
Responder states: MQTT(0) HTTP(0) HTTP(0) HTTP(0) HTTP(0) FTP(0) Telnet(0)
HTTP sessions: 1 of 8
= WiFi =
Interface state: active
Module is connected to access point
Failed messages: pending 0, notready 0, noresp 0
Firmware version 1.26
MAC address f0:08:d1:03:81:a2
Module reset reason: Power up, Vcc 3.38, flash size 2097152, free heap 24792
WiFi IP address 192.168.32.177
Signal strength -45dBm, channel 0, mode 802.11n, reconnections 0
Clock register 00002002
Socket states: 0 0 0 0 0 0 0 0
```

### What happened?

Prints do not always complete. The same print sometimes completes and sometimes does not. The head may be moved away from the print, or it may get left on the print. Some heaters may be turned off.

Observed by two forum users, see forum link. Also observed by dc42 on toolchanger.

dc42, 2023-08-31:
May be fixed by the fix for issue #901.

tristanryerparke, 2023-09-20:
I've also noticed this issue on my 6HC + 4x 1HCL machine running 3.5.0rc1

dc42, 2023-10-06:
Fixed the issue that caused the File channel sometimes to remain in sync state 1 when the print completed. This may or may not fix the original issue.

dc42, 2023-10-11:
Fixed the issue with print jobs on the delta sometimes ending up with the File channel stuck in a sync state. This will likely fix the other issue too.

dc42, 2023-11-13:
Confirmed fixed by forum users, see original forum thread.

closed 2023-11-13 by dc42