Duet3D/RepRapFirmware #900: [FeatureRequest]: Add support for LIS2DW12 Accelerometer

jaysuk, 2023-08-30:
### Is your feature request related to a problem? Please describe.

Now klipper has support for the LIS2DW12, it would make sense for accelerometer support to be extended to include LIS2DW12 support. Here is the information https://www.st.com/en/mems-and-sensors/lis2dw12.html
Here's the klipper PR https://github.com/Klipper3d/klipper/pull/6312

### Describe the solution you propose.

Add LIS2DW12 support

### Describe alternatives you've considered

none

### Provide any additional context or information.

none

jaysuk, 2023-09-03:
Its has a slightly smaller package size (2 x 2 mm) than the LIS3DH (3 x 3 mm)
It has a lower noise than the LIS3DH.
The only real limiting factor is that it has a lower max data output rate (1600 Hz) compared to the LIS3DH (5300 Hz).
If we could get support it would mean the only thing stopping us having a universal toolboard between RRF and Klipper is the requirement for the MCP2518FD but I'm trying to find someone to sort that as the MCP can also handle CAN 2.0 data (https://klipper.discourse.group/t/add-support-for-mcp2518fd-for-rp2040-based-toolboards-looking-for-a-dev/10210) (that or we have some sort of jumper arrangement so the end user can select the type of connection).

On another note, I notice the LIS3DH has an internal temperature sensor. Would it be possible (or any use) to make this available to RRF?

dc42, 2023-11-13:
Implemented in 3.5-dev RRF and Duet3Expansion source code.

closed 2023-11-13 by dc42