Duet3D/RepRapFirmware #909: [Bug]: RepRapFirmware 3.5.0-rc.1 and Screens 12864

X-Dron, 2023-09-13:
### Duet Forum Discussion Thread

https://forum.duet3d.com/categories

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

RepRapFirmware 3.5.0-rc.1

### Duet Web Control Version

DWC 3.5.0-rc.1

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [ ] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122 Report
```

### Please upload the content of your config.g file.

```gcode
Config.g
```

### What happened?

Hi, after these changes:
M98 now requires the P parameter to be a quoted string. Previously, in standalone mode an unquoted string was permitted.
M98 now always requires the P parameter to be a quoted string. 
starting macros from screens 12864 does not work
Does not work:
button R0 C0 W63 F0 H1 T"Auto Level" A"M98 P0:/menu/scripts/autoLevel.g" 
Does not work too:
button R0 C0 W63 F0 H1 T"Auto Level" A"M98 P#0" L"/menu/scripts/autoLevel.g"
In RRF 3.5.0-beta.4 everything works correctly

T3P3, 2023-09-13:
can you try using:

button R0 C0 W63 F0 H1 T"Auto Level" A""M98 P0:/menu/scripts/autoLevel.g""

X-Dron, 2023-09-13:
Does not work (((

T3P3, 2023-09-13:
apologies, try

button R0 C0 W63 F0 H1 T"Auto Level" A """M98 P0:/menu/scripts/autoLevel.g"""

dc42, 2023-09-13:
Should be this I think:
```
button R0 C0 W63 F0 H1 T"Auto Level" A"M98 P""0:/menu/scripts/autoLevel.g"""
```

oliof, 2023-09-13:
> Does not work (((
> There is an empty command button and the command is not executed.

that's missing a third " between P and 0 ...

-------- Original-Nachricht --------

X-Dron, 2023-09-13:
> that's missing a third " between P and 0 ... Gesendet von ProtonMail mobile
> […](#)

Does not work (((
_________________________
Error loading menu
File: bedLevel
Line 21 collum 48
Bad arg letter
________________________
See src/Display/Menu.cpp
const char *Menu::ParseMenuLine(char * const commandWord) noexcept

X-Dron, 2023-09-14:
I think the best way is to change  
ButtonMenuItem::Select(const StringRef& cmd)
in 
/src/Display/ButtonMenuItem.cpp

```
bool ButtonMenuItem::Select(const StringRef& cmd) noexcept
{
	const int nReplacementIndex = StringContains(command, "#0");
	if (-1 != nReplacementIndex)
	{
		cmd.copy(command, nReplacementIndex);
		cmd.cat(m_acFile);
	}
	else
	{
		cmd.copy(command);
		if (StringEqualsIgnoreCase(command, "menu") && strlen(m_acFile) != 0)
		{
			// For backwards compatibility, if 'menu' is used without any parameters, use the L parameter as the name of the menu file
			cmd.cat(' ');
			cmd.cat(m_acFile);
		}
	}

	return true;
}

```

dc42, 2023-09-15:
I will look at this in more detail next week, but my current plan is to change ParseMenuLine to recognise a pair of adjacent double quote characters within a string literal as representing and embedded double quote character.

X-Dron, 2023-09-17:
> I will look at this in more detail next week, but my current plan is to change ParseMenuLine to recognise a pair of adjacent double quote characters within a string literal as representing and embedded double quote character.

I think it's better to leave the menu file syntax as is. Thus, there is no need to update the parser. It parses the command and file path correctly. The G-code command generation syntax needs to be updated.

dc42, 2023-09-24:
The problem lies in the parser. Given the argument A"M98 P""test.g""" it sets the string argument to"M98 P", then it parses "test.g" and treats that as an implicit T parameter (changing the button text), then it parses the final "" and treats that as an implicit T parameter (changing the button text to an empty string). So the parser needs to be fixed.

RRF in standalone mode used to allow the P argument of M98 not to be quoted e.g. A"M98 Ptest.g" but that causes problems if you want to pass parameters to the macro.

I have fixed the parser so that a line such as this:

`button C64 T" Macro " A"M98 P""test.g"""
`
now works.

closed 2023-09-24 by dc42

pfn, 2023-11-15:
@dc42 This is a bad solution IMO as it introduces inconsistency into the menus language, e.g. for the `files` command `A"M98 P#0"` does not need the `#0` argument quoted, whereas for `button` it is necessary. Has the change been made in a way that `files` must also be modified such that `#0` is quoted?

dc42, 2023-11-15:
#0 expands to a quoted string, therefore it does not need to be quoted.

pfn, 2023-11-16:
It (`#0`) isn't currently working in rc1, so it is fixed in rc1+?