Duet3D/RepRapFirmware #915: [FeatureRequest]: Add "suppress message" G30 probe option

Braintoe, 2023-09-26:
### Is your feature request related to a problem? Please describe.

IDEX machines are rather cumbersome to calibrate without some support skript. While RepRepFirmware brings all the features needed to create such a skript which can then display comprehensive leveling instructions (e.g. an ASCII bar graph-like representation, see [here in the Duet forum](https://forum.duet3d.com/topic/32657/displaying-z-level-instructions-on-paneldue/5) what I mean) with conditional gcode, the G30 probe function itself prevents such a custom message by displaying a message for every single probe move, regardless which option was chosen. If continuous probing is done, this means the display (PanelDue or Web) constantly flickers between the G30 message and the user message.

Additionally, this behaviour is not described in the Gcode dictionary. the text there states that only a G30 S-1 move creates a message.

### Describe the solution you propose.

There are two options to solve this:
- either add a new parameter to switch the message on or off. This would be a universal solution since it could be combined with all S parameters.
- or add an S parameter "Probe the bed at the current XY position. When the probe is triggered, do not adjust the Z coordinate, just update the sensors.probes[].lastStopHeight value in the object model" that behaves like S-1 but suppresses the corresponding message.

### Describe alternatives you've considered

Workarounds considered:
- using some "S-" parameter of G30. Sadly, they all create messages.
- abusing a G38.2 probe towards workpiece move. For some reason, this probe move had significant issues (firmware 3.5.0rc1) - it was between 10 and 30 times less accurate, and it kept moving the X axis together with the Z axis without even updating the X axis position, while every other move worked perfectly. See https://forum.duet3d.com/topic/33675/g38-2-much-less-precise-than-g30-and-moving-wrong-axes-bug for details
- recreating a "no-message-G30" move with conditional gcode: "while sensors.probes[1].value[0] < sensors.probes[1].threshold...". Does not seem to work unless you move in 0.01mm steps or less which takes much too long - as I understand it because the Duet Gcode channel system checks the while condition before executing the loop, not while it is executed.
- temporarily redefining the z probe as z endstop and using a G1 H2 command to find out the contact point. Deemed too risky since the z probe contacts on my printer require the nozzle to be at a defined position which is not the homing position, with the print plate removed, otherwise they won't work at all. Thus, if one of the macros fails for any reason, the next z homing move will end in a damaged printer (the normal z endstop is on the high end of z, while a z probe will contact on the low end of z)

That's about it. I am open for suggestions ;-)

### Provide any additional context or information.

Ex. Photos, mockups, etc.

Braintoe, 2023-09-26:
(Edit: completely forgot - in any case, update the documentation of G30 that others who might try this on similar Marlin-based printers don't only find out _after_ they bought the Duet...)

dc42, 2023-09-27:
@Braintoe please clarify: 

1. Which particular form of the G30 command are you referring to? AFAIR a plain G30 command with no parameters does not result in a message; but G30 S-1 does.
2. Please provide an example of the sort of script you want to use, where the messages produced by G30 commands in that script are undesirable.

Braintoe, 2023-09-28:
@dc42 I tested G30 S-2, G30 S-3 and G30 - all created a message of some sorts: the first two ones created a notification, the last one complained about someting (the amount of adjustment screws?). What I actually want, is a G30 S-1 witout a message that I can use in a conditional gcode script, but I can save and reset the changes that G30 S-2 and G30 S-3 make if needed.

Below are the scripts I want to use. the first one 
1. probes z several times at the first bed level adjustment screw, then updates z max and the current position. 
2. After that, it continuously probes z on the other two adjustment screws and notifies the user with messages on the PanelDue how to turn the adustment screws on the bed (due to the constraints of the PanelDue I made it an ASCII art bar graph which works well). When the script is finished, the bed is leveled and the z distance to hotend 1 is correct.
The first file is the original code with german comments which is partially tested by now (tested until the "TODO" line, and the message generation is tested by replacing the G30 codes with varying dummy data. The second one is an automated translation into english for your convenience. I did my best to correct the variable names in the translation (and will probably never use Deepl to translate such code again), but please do not expect it to work...

Basically the same type of script is also used to z-level hotend 2 of the IDEX printer against hotend 1 - hotend 1 does a z probe which is definded as reference, then hotend 2 continuously probes at the same spot, and the user sees the same messages and can trim the z level of the right hotend by turning an adjustment screw.

Afterwards, just an XY calibration needs to be done which is done by using the wonderful M675 gcode in a cavity with both hotends and calculating the XY offset of hotend 2 from those measurements..
[auto_bedleveling_english_translation.g.txt](https://github.com/Duet3D/RepRapFirmware/files/12744700/auto_bedleveling_english_translation.g.txt)
[auto_bedleveling_german_comments.g.txt](https://github.com/Duet3D/RepRapFirmware/files/12744701/auto_bedleveling_german_comments.g.txt)