Duet3D/RepRapFirmware #919: DHT22 sensor glitch

T3P3, 2023-10-04:
DHT22 sensors sometime report 2000C for both humidity and temperature after running for a period of time. Undefining and redefining them in RRF does not fix this. Unplugging them and plugging them back in does.

Discovered on Mini5+ and Duet2. On the mini5+ the DHT Sensor is wired to an IO pin on the mainboard (+ GND and the IN/OUT pins bridged) for the duet 2 it is wired to the connector for the temperature daughterboard.

dc42, 2024-01-15:
This is caused by noise on the one-wire line causing the DHT22 to lock up. A change made by chrishamm to hold the wire high when the sensor is not active instead of relying solely on the pullup resistor has improved it but not fixed it. We could perhaps improve it further by pulling the line high in the ISR when sufficient pulses have been received, rather than after waiting for a period of time sufficient to receive all the pulses. If we do this, we may as well also do the data decoding in the ISR (i.e. decode into 0 and 1 bits) instead of capturing it in the ISR and decoding it late, then we wouldn't need the capture buffer.

T3P3, 2024-03-06:
Some noise will lockup the DHT22 sensor.  This is an issue with all DHT22 sensors but can be more prevalent in clones.

A pullup resistor is recommended in longer/nosier cables (e.g. 4K7) 

Shielded cables will also help in some situations

closed 2024-03-06 by T3P3