Duet3D/RepRapFirmware #920: MFM not working in 3.5rc1

T3P3, 2023-10-12:
Reported by OEM and on forum:
https://forum.duet3d.com/topic/33822/filament-monitor-not-working

dc42, 2023-11-01:
Fixed various issues with MFMs (they were not working at all when a main board was used as an expansion board, however the customer reports that the issue remains. New firmware build with additional diagnostics made available to customer.

dc42, 2023-11-13:
The particular issue is that some time into the print, the filament monitor stops accumulating total extrusion and stops comparing measured vs. accumulated extrusion. The M122 report shows that data reception errors have occurred; whereas on my delta that uses a MFM attached to a Duet 3 Mini, the filament monitor doesn't stop working and I see no data reception errors. So perhaps the issue is failure to recover from a data reception error.

dc42, 2023-11-13:
I asked for two M122 and M409 reports to be taken after the filament monitor stop accumulating extrusion, a few minutes apart.

Reports received 31 Oct 23 show:
 totalExtrusion in the OM: stuck at 206.9
 first M122: Extruder 0: pos 13871.95, errs: frame 0 parity 2 ovrun 0 pol 2 ovdue 0
 second M122: Extruder 0: pos 13907.81, errs: frame 2 parity 2 ovrun 0 pol 3 ovdue 0.

Reports received 2 nov 23 show:
totalExtrusion: stuck at 225.6
first M122: check 5724205 clear 0 Extruder 0: pos 2318.20, errs: frame 9 parity 0 ovrun 0 pol 3 ovdue 0
second M122 check 5943297 clear 0 Extruder 0: pos 13902.19, errs: frame 9 parity 0 ovrun 0 pol 3 ovdue 0

dc42, 2023-11-20:
Report received 17nov23:
1st M122 at 08:06:09 : in 9299 notIn 2121025 Extruder 0: pos 14024.88, debug 9269 8603 8498 7908 -174530.66, errs: frame 32 parity 0 ovrun 0 pol 6 ovdue 0
2nd M122 at 08:13:45: in 19430 notIn 3439590 Extruder 0: pos 13997.11, debug 19401 18734 18502 17410 -173350.59, errs: frame 32 parity 1 ovrun 0 pol 8 ovdue 0

closed 2024-01-26 by T3P3