Duet3D/RepRapFirmware #924: Potential issue with delta printers

T3P3, 2023-10-25:
https://forum.duet3d.com/topic/33886/y-slope-with-3-5-0-rc1-not-seen-with-3-4-6/14

dc42, 2023-11-01:
User reports it is solved by the latest 3.5.0-rc1+ build, see forum thread.

dc42, 2024-04-05:
Issue is not solved and affects 3.5.0-rc.3. See https://forum.duet3d.com/post/335641 and the following posts.

dc42, 2024-04-05:
Found two causes:
1. We were only allowing step times to be calculated late (beyond the move duration) in the last segment of the move. On delta printers we have previously seen that the last two steps may be calculated late. When input shaping is enabled the last segment may contain only one step, so the penultimate segment may have a late step calculation. This should have led to a step error being logged, but the code to record step errors was not working because the error status was being cleared when the move was seen to have completed (or in this case, aborted).
2. A delta move may sometimes have zero net steps but a DM is still created in case the carriage needs to move up and then down by the same amount during the move. However, if the move really didn't require any steps then although totatSteps was set to zero one step was always taken.

dc42, 2024-04-10:
Confirmed fixed.

closed 2024-04-10 by dc42