Duet3D/RepRapFirmware #925: Return full code reply via HTTP rr_reply code for corresponding rr_gcode calls

chrishamm, 2023-10-26:
@T3P3 requested a better `rr_reply` HTTP interface which returns the matching G-code reply of a previously issued `rr_gcode` request. At the moment DWC replies can be duplicated  (and logged multiple times, e.g. in a macro when dealing with `M291`/`M292` and `echo` calls) which makes code reply parsing in plugins more difficult and has the potential to confuse users.

Possibly `rr_gcode` could return the next sequence number and `rr_reply` could accept that previously returned sequence number to return only a subset of the buffered G-code reply. `M122` and other codes that do subsequent Platform::Message(F) would need special handling, though.

T3P3, 2025-10-26:
one of the use cases for this:
https://forum.duet3d.com/topic/34422/how-to-preserve-message-order-in-the-web-console

T3P3, 2026-09-12:
Forum threads:
https://forum.duet3d.com/topic/37985
https://forum.duet3d.com/topic/31420