Duet3D/RepRapFirmware #926: [Bug]: M954 by itself doesn't output the set CAN address

jaysuk, 2023-11-03:
### Duet Forum Discussion Thread

N/A

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [X] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.4.6

### Duet Web Control Version

3.4.6

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
N/A
```

### Please upload the content of your config.g file.

```gcode
N/A
```

### Please upload the content of any other releveant macro files.

```gcode
N/A
```

### Details specific to your printer.

N/A

### Links to additional info.

N/A

### What happened?

It's expected that when a Gcode that doesn't have any arguements outputs it's assigned value
M954 doesn't and it expects Ax to also be included

dc42, 2024-02-19:
What would be the purpose of this? The CAN address of the main board is always zero, and to use M954 on an expansion board you need to specify its CAN address using the B parameter. Either way, you already know the CAN address. The only situation in which I think it has a possible use is if you configure a main board as an expansion board, and then send commands directly to that board over USB or UART; but that's an odd use case.

T3P3, 2024-05-20:
not planned until we understand a usecase.

closed 2024-05-20 by T3P3