Duet3D/RepRapFirmware #935: Regular crashes using 3.5.0-rc.1

dc42, 2023-11-13:
See thread https://forum.duet3d.com/topic/33998/reboots-crashes-rrf-3-5-0-rc1. Only one user is currently reporting this, however there are some similarities in the stack trace each time which suggests that it may be a firmware issue.

dc42, 2023-11-13:
Common factors in stack traces:
- Task is always NETW (aka NETWORK)
- LR is 0009e9cd in the last 3 reports (running rc1+), 0009dff9 and 0009df2d in the first two (running rc1)
- Other stack values that look like code: 0002ff97 in the last 3, 0002fedf in the first two

dc42, 2023-11-14:
Added check for return address corruption in WiFiInterface::SendCommand. Now waiting for user to test it and report, see https://forum.duet3d.com/post/327312.

dc42, 2023-12-07:
Return address corruption, when it occurs, occurs during the call to Disable Spi(). The data written appears to be a single 32bit word and is always either zero or what looks like 4 ASCII characters. Character patterns seen are `=0.1` and `=0.6` . I have added a debug watchpoint on the stored return address to see if it is the processor core that writes to it. Now waiting for the user to test it.

dc42, 2024-05-14:
These reports https://forum.duet3d.com/topic/35611/my-printer-reboots-alone-3-5-1/ look to me like instances of the same issue although the stack location that is overwritten by zeros is different.

dc42, 2024-05-30:
Another likely instance: https://forum.duet3d.com/topic/35833/duet-3-mini5-nightly-restarts/

droftarts, 2024-07-22:
Another possible instance: https://forum.duet3d.com/topic/36094/duet-3-mini-5-standalone-3-5-2-hardfault-bfarvalid-precise

dc42, 2024-08-12:
Another possible instance https://forum.duet3d.com/topic/36314/hardfault-imprec-hardfault-invstate-printer-crashes

droftarts, 2024-09-17:
Another possible instance https://forum.duet3d.com/topic/36592/duet-3-mini5-randomly-resets-during-printing

pfn, 2024-10-20:
Also just got this today using 3.5.3:

```
Last reset 00:00:31 ago, cause: software
Last software reset at 2024-10-19 20:57, reason: HardFault invState, Gcodes spinning, available RAM 404, slot 1
Software reset code 0x4063 HFSR 0x40000000 CFSR 0x00020000 ICSR 0x00000803 BFAR 0xe000ed38 SP 0x20012008 Task NETW Freestk 482 ok
Stack: 0000017e 00000002 200014e4 00000000 20035d6a 000a0f9f 00000000 600f0000 00000000 00000000 00000000 00000000 200343a4 00000800 200388d0 2002c640 2001882c 2002c496 2001882c 2001ea80 00030523 00000000 00000000 00000000 200120b8 00000014 ffffffff
Error status: 0x00
```

On my mini5+2 (clone) w/ 2 CAN-FD boards (exp3 hc & sb2040 picocan)

dc42, 2024-10-22:
@Perry please install the 3.5.3+1 firmware linked from post
https://forum.duet3d.com/post/346327.

dc42, 2024-11-26:
Fixed in 3.5.4.

closed 2024-11-26 by dc42