Duet3D/RepRapFirmware #940: M291 Multi-line

myfordvmb, 2024-01-03:
### Is your feature request related to a problem? Please describe.

Would like the ability to have a multi-line message in M291.
Propose \n for new line.
This is to format information to user.

### Describe the solution you propose.

M291 P{"Variable A="^global.VarA^\n^Variable B="^global.VarB} S2
would result in Message Dialog
Variable A=5
Variable B= 10

### Describe alternatives you've considered

Tried echo but goes to console and does not prompt user.

### Provide any additional context or information.

Ex. Photos, mockups, etc.

dc42, 2024-01-05:
Another possible implementation would be for the M291 P parameter to accept an array of strings.

T3P3, 2026-09-16:
Forum threads:
https://forum.duet3d.com/topic/34578
https://forum.duet3d.com/topic/32657