Duet3D/RepRapFirmware #942: 1LC - stepper randomly reversing - ≤3.5.0-r1

T3P3, 2024-01-10:
As described in this thread:

https://forum.duet3d.com/topic/34351/1lc-stepper-randomly-reversing-3-5-0-r1

Exerqtor, 2024-01-10:
I've ran upwards to 150hours of prints on 3.5.0-beta4 now without anything happening. So I'm quite positive the issue causing this behaviour got implimented as of rc1.

dc42, 2024-03-13:
Issue was caused when moves with zero duration were received. This resulted in a floating point division by zero, which ended up with segments being created with infinite length, causing the extrusion distance brought forwards being infinite. Most likely the behaviour when zero length messages were received changed when the floating point maths library was switched to Qfplib which happened between 3.5 beta4 and rc1.

The zero duration moves probably occur because moves generated by the main board have the durations of the acceleration. steady and deceleration segments rounded to an integer number of step clocks to send over the CAN message. So a move with very small duration may get rounded to zero duration.

Fixed by checking for zero duration moves when they are received and changing the duration of the steady speed segment to 1 clock when they occur. Doing this ensures that any small amount of extrusion requested will get added to the extrusion pending. Also changed the way that the acceleration and deceleration distances are computed, to force them to zero if the corresponding numbers of clocks are zero, guarding against possible miscalculation (e.g. due to rounding error) if the full calculation is done.

closed 2024-03-15 by dc42