Duet3D/RepRapFirmware #947: [FeatureRequest]: Add Slicer Print time to JSON Resopnse

SirBramble, 2024-01-23:
### Is your feature request related to a problem? Please describe.

I've recently integrated my Printer (Duet 2 recently upgraded to Firmware 3.4.6) in to Home Assistant and am using the JSON response to get the current print times, temperatures, etc. from the current Print.

However the remaining print times based on file/filament/layer vary quite a lot and are inaccurate at the beginning to middle of a print.

### Describe the solution you propose.

It would be great if the print time based on the slicer, shown in the Printers 'Job/Status' page, would be included in the JSON response.

### Describe alternatives you've considered

I have not found a workaround so far.

### Provide any additional context or information.

[image]

I'm also a bit confused, because I found the parameter with the Object model plugin:
[image]

But I'm also new to all of this, so if I'm just missing something obvious, help would be greatly appreciated.

T3P3, 2024-01-23:
The slicer time left should be in the OM (e.g.: here):
https://github.com/Duet3D/RepRapFirmware/wiki/Object-Model-Documentation-Beta-&-RC#jobtimesleftslicer

But AFAIK the slicer needs to set the time.

SirBramble, 2024-01-23:
Thanks for the quick reply.

I found it in the object model, but it isn't included in the JSON response (at least not as far as I can tell).

This is my current YAML config:
```
- platform: rest
    name: Duet Status
    resource: http://192.168.0.52/rr_status?type=3
    value_template: "{{ value_json.status }}"
    force_update: true
    json_attributes:
      - temps
      - currentLayer
      - fractionPrinted
      - printDuration
      - timesLeft
  - platform: template
    sensors:
      print_duration:
        friendly_name: "Time Printed"
        value_template: "{{ states.sensor.duet_status.attributes.printDuration // 60 }}"
        unit_of_measurement: "min"
      print_timeleft:
        friendly_name: "Est Time Left"
        value_template: "{{ states.sensor.duet_status.attributes.timesLeft.slicer // 60 }}"
        unit_of_measurement: "min"
```
"Time Printed" works, "EST Time Left" does not
[image]

Am I missing something here?

dc42, 2024-01-23:
It works for me, using PrusaSlicer and RRF 3.5.0-rc.2+.
[image]

dc42, 2024-01-23:
If you are using the M408 response or the rr_status response, be aware those are legacy features. They will nt be extended and they will be removed in a future version, perhaps as soon as RRF 3.6.0.

SirBramble, 2024-01-23:
Ok, Interesting. I'm on 3.4.6 and got this response:
[image]
[image]
Sorry for the bad picture, but I have no other way to show the values.

I only have file and filament.

For reference:
URL called: http://192.168.0.52/rr_status?type=3
Slicer: Cura V5.4.0
Showes up in the Interface:
[image]

Is there maybe a setting or config option I have to change?

SirBramble, 2024-01-24:
Ok, so I got it working with the object model. Still need to port the rest of the parameters from rr_status, but at least it workes now.

@dc42 Thanks for the Help. The M408 gcode showed me in the right direction.

For anyone stumbling across this:
YAML config:
```
- platform: rest
    name: Duet Status
    resource: http://192.168.0.52/rr_status?type=3
    value_template: "{{ value_json.status }}"
    force_update: true
    json_attributes:
      - temps
      - currentLayer
      - fractionPrinted
      - printDuration
  - platform: rest
    name: Duet Status new
    resource: http://192.168.0.52/rr_model?key=job&flags=n
    value_template: "{{ value_json.key }}"
    json_attributes_path: "$.result"
    force_update: true
    json_attributes:
      - timesLeft
      - lastDuration
      - duration
  - platform: template
    sensors:
      head_temp_current:
        friendly_name: "Nozzle Temperature Current"
        value_template: "{{ states.sensor.duet_status.attributes.temps.current[1] }}"
        unit_of_measurement: "°C"
      head_temp_active:
        friendly_name: "Nozzle Temperature Active"
        value_template: "{{ states.sensor.duet_status.attributes.temps.tools.active[0][0] }}"
        unit_of_measurement: "°C"
      bed_temp_current:
        friendly_name: "Bed Temperature Current"
        value_template: "{{ states.sensor.duet_status.attributes.temps.current[0] }}"
        unit_of_measurement: "°C"
      bed_temp_active:
        friendly_name: "Bed Temperature Active"
        value_template: "{{ states.sensor.duet_status.attributes.temps.bed.active }}"
        unit_of_measurement: "°C"
      print_percentage:
        friendly_name: "Percentage Printed"
        value_template: "{{ states.sensor.duet_status.attributes.fractionPrinted }}"
        unit_of_measurement: "%"
      print_duration:
        friendly_name: "Time Printed"
        #value_template: "{{ states.sensor.duet_status_new.attributes.duration // 60.0|float(0.0)}}"
        value_template: "{{ state_attr('sensor.duet_status_new', 'duration')|int(0) // int(60) }}"
        unit_of_measurement: "min"
      print_timeleft:
        friendly_name: "Est Time Left"
        value_template: "{{ states.sensor.duet_status_new.attributes.timesLeft.slicer|int(0) // int(60)}}"
        unit_of_measurement: "min"

homeassistant:
  customize:
    sensor.duet_status:
      friendly_name: "3D Printer Status"
      icon: mdi:printer-3d

group:
  3dprinter:
    name: Ender5
    entities:
      - sensor.duet_status
      - sensor.duet_status_new
      - sensor.head_temp_current
      - sensor.head_temp_active
      - sensor.bed_temp_current
      - sensor.bed_temp_active
      - sensor.print_percentage
  3dprintstats:
```
Plugin for model browser (useful to look up parameter location):
[image]

Useful links:
[https://github.com/Duet3D/RepRapFirmware/wiki/HTTP-requests#get-rr_model](https://github.com/Duet3D/RepRapFirmware/wiki/HTTP-requests#get-rr_model)
[https://docs.duet3d.com/User_manual/Reference/Gcodes](https://docs.duet3d.com/User_manual/Reference/Gcodes) (under M409)
[https://jsonviewer.stack.hu/](https://jsonviewer.stack.hu/) (useful for visualizing JSON response)
[http://LOCAL_IP_PRINTER/rr_model?key=job&flags=n](http://LOCAL_IP_PRINTER/rr_model?key=job&flags=n) (get objectModel/Job JSON response from browser)

T3P3, 2024-01-24:
thanks for the detailed reply.

closed 2024-01-24 by T3P3