Duet3D/RepRapFirmware #953: [Bug]: Releases after 2.02 (RepRapFirmware.bin 1.23) broken calibration for delta on duet 0.6

casesolved-co-uk, 2024-01-29:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/34823/tower-clash-in-mesh-bed-compensation-fisher-delta

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [X] Other
- [ ] None

### Firmware Version

FIRMWARE_NAME: RepRapFirmware for Duet FIRMWARE_VERSION: 1.23 ELECTRONICS: Duet 0.6 FIRMWARE_DATE: 2018-12-24b1

### Duet Web Control Version

Web Interface Version:	1.22.6

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
=== Diagnostics ===
RepRapFirmware for Duet version 1.23 running on Duet 0.6
Used output buffers: 3 of 16 (10 max)
=== System ===
Static ram: 44276
Dynamic ram: 42884 of which 2952 recycled
Stack ram used: 136 current, 3392 maximum
Never used ram: 4800
=== Platform ===
Last reset 00:09:10 ago, cause: software
Last software reset at 2024-01-29 13:15, reason: User, spinning module GCodes, available RAM 5112 bytes (slot 2)
Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x0400f000 BFAR 0xe000ed38 SP 0xffffffff
Error status: 0
Free file entries: 10
SD card 0 detected, interface speed: 21.0MBytes/sec
SD card longest block write time: 0.0ms, max retries 0
MCU temperature: min 46.5, current 48.0, max 48.6
Date/time: 2024-01-29 13:33:22
Slowest loop: 60.32ms; fastest: 0.09ms
I2C nak errors 0, send timeouts 0, receive timeouts 0, finishTimeouts 0
=== Move ===
Hiccups: 15, StepErrors: 0, LaErrors: 0, FreeDm: 100, MinFreeDm: 97, MaxWait: 35080ms, Underruns: 0, 0
Scheduled moves: 47, completed moves: 47
Bed compensation in use: mesh
Bed probe heights: 1.821 5.044 0.416 0.942 0.000
=== Heat ===
Bed heaters = 0, chamberHeaters = -1 -1
=== GCodes ===
Segments left: 0
Stack records: 2 allocated, 0 in use
Movement lock held by null
http is idle in state(s) 0
telnet is idle in state(s) 0
file is idle in state(s) 0
serial is idle in state(s) 0
aux is idle in state(s) 0
daemon is idle in state(s) 0
queue is idle in state(s) 0
autopause is idle in state(s) 0
Code queue is empty.
=== Network ===
Free connections: 15 of 16
Free transactions: 23 of 24
Locked: 0, state: 4, listening: 20071c18, 0, 0
```

### Please upload the content of your config.g file.

```gcode
; Configuration file for Duet 0.6 (firmware version 1.21)
; executed by the firmware on start-up
;
; generated by RepRapFirmware Configuration Tool v3.4.0 on Mon Jan 29 2024 11:54:44 GMT+0000 (Greenwich Mean Time)

; General preferences
G90                                         ; send absolute coordinates...
M83                                         ; ...but relative extruder moves
M550 P"MenShed"                             ; set printer name
; config-override.g will override these after calibration (M500) with M501 below
M665 R78 L160 B75 H162                      ; Set delta radius, diagonal rod length, printable radius and homed height
M666 X0 Y0 Z0                               ; put your endstop adjustments here, or let auto calibration find them

; Network
;M551 P"MenShed"                             ; set password - used for all protocols inc HTTP
M540 PBE:EF:DE:AD:FD:05                     ; set custom MAC address
M552 P0.0.0.0 S1                            ; enable network and acquire dynamic address via DHCP
M586 P0 S1                                  ; enable HTTP
M586 P1 S0                                  ; disable FTP
M586 P2 S0                                  ; disable Telnet

; Drives
M569 P0 S1                                  ; physical drive 0 goes forwards
M569 P1 S1                                  ; physical drive 1 goes forwards
M569 P2 S1                                  ; physical drive 2 goes forwards
M569 P3 S1                                  ; physical drive 3 goes forwards
M584 X0 Y1 Z2 E3                            ; set drive mapping
M92 X87.49 Y87.49 Z87.49 E137.00            ; set steps per mm
M566 X1200.00 Y1200.00 Z1200.00 E1200.00    ; set maximum instantaneous speed changes (mm/min)
M203 X15000.00 Y15000.00 Z15000.00 E9000.00 ; set maximum speeds (mm/min)
M201 X4000.00 Y4000.00 Z4000.00 E4000.00    ; set accelerations (mm/s^2)
M906 X800 Y800 Z800 E1000 I30               ; set motor currents (mA) and motor idle factor in per cent
M84 S30                                     ; Set idle timeout

; Axis Limits
M208 Z-0.2 S1                               ; set minimum Z

; Endstops
M574 X2 Y2 Z2 S1                            ; set active high endstops

; Z-Probe
M558 P4 H5 F120 T12000                      ; set Z probe type to switch and the dive height + speeds
G31 P500 X0 Y0 Z-0.1                        ; set Z probe trigger value, offset and trigger height
M557 R65 S20                                ; define mesh grid

; Heaters
;M140 H-1                                    ; disable heated bed
M305 P0 R4700 T100000                       ; Uncomment for heatbed
M143 H0 S110                                ; set temperature limit for heater 0 to 110C
M305 P1 T100000 B4388 R4700                 ; set thermistor + ADC parameters for heater 1
M143 H1 S280                                ; set temperature limit for heater 1 to 280C

; Fans
M106 P0 S0 I0 F500 H-1                      ; set fan 0 value, PWM signal inversion and frequency. Thermostatic control is turned off

; Tools
M563 P0 S"Nozzle" D0 H1 F0                  ; define tool 0
G10 P0 X0 Y0 Z0                             ; set tool 0 axis offsets
G10 P0 R0 S200                              ; set initial tool 0 active and standby temperatures to 200C/0C

; Custom settings
M301 H1 P5.3 I0.368 D14.2 W180 B300         ; Set nozzle heater PID parameters
M501                                        ; read axis and heater calibration settings from config-override.g. Overrides above M665 and M666 settings
M570 H1 P20 T15                             ; relax the heater fault causing bricked prints! 15 degrees for 20 seconds
```

### Please upload the content of any other releveant macro files.

```gcode
bed.g

; bed.g
; called to perform automatic delta calibration via G32
;
; generated by RepRapFirmware Configuration Tool v3.4.0 on Mon Jan 29 2024 11:54:44 GMT+0000 (Greenwich Mean Time)
M561 ; clear any bed transform
; Probe the bed at 3 peripheral and 0 halfway points, and perform 3-factor auto compensation
; Before running this, you should have set up your Z-probe trigger height to suit your build, in the G31 command in config.g.
G30 P0 X-56.21 Y-32.45 H0 Z-99999           ; X tower
G30 P1 X56.21 Y-32.45 H0 Z-99999            ; Y tower
G30 P2 X0 Y64.9 H0 Z-99999                  ; Z tower
G30 P3 X0 Y0 H0 Z-99999 S3                  ; centre
; Use S-1 for measurements only, without calculations. Use S4 for endstop heights and Z-height only. Use S6 for full 6 factors
; If your Z probe has significantly different trigger heights depending on XY position, adjust the H parameters in the G30 commands accordingly. The value of each H parameter should be (trigger height at that XY position) - (trigger height at centre of bed)

G29 S1                                ; load the saved height map from mesh bed compensation; must be run AFTER G32
```

### Details specific to your printer.

Duet 0.6 Reprap Fisher Delta

### Links to additional info.

https://forum.duet3d.com/topic/34820/reprap-fisher-latest-firmware-for-duet-0-6

### What happened?

On firmware versions after 1.23 (2.02) (I have tried all versions: 1.24, 1.25, 1.26 & 1.26.1 i.e 2.03 to 2.05.1) either bed calibration (G32) or mesh bed calibration (G29 S0) have a faulty motion sequence.
The first thing they do is move toward the Z tower at various speeds.
1.26.1 clashes the Z tower.

For firmware 1.24-1.26 bed calibration can be done without Z tower clash but the sequence is something like:

### Expected result

As 1.23:

* Do some slow movement
* Move to centre
* Move to probe point
* Probe

### Observed result

* Move a long way from centre ?
* Do some slow movement
* Move to centre
* Move to probe point
* Probe
* Move a long way from centre ?

### Steps to reproduce

Run bed calibration G32

dc42, 2024-01-29:
Check that you don't have files called deployprobe.g or retractprobe.g in /sys on the SD card.

chrishamm, 2024-01-29:
Fixed in legacy configtool version 3.4.1.

casesolved-co-uk, 2024-01-29:
> Check that you don't have files called deployprobe.g or retractprobe.g in /sys on the SD card.

That appears to have fixed it! Thanks.
Now testing with 1.26.1

closed 2024-01-29 by droftarts