Duet3D/RepRapFirmware #954: [Bug]: M122 causing Lost connection to Duet

timschneider, 2024-02-04:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/34838/3-5-0-rc-3-m122-causing-lost-connection-to-duet-timeout

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [X] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.5.0-rc.3

### Duet Web Control Version

3.5.0-rc.3

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [X] Yes I use a SBC.
- [ ] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
=== Diagnostics ===
RepRapFirmware for Duet 3 MB6HC version 3.5.0-rc.3 (2024-01-24 17:58:49) running on Duet 3 MB6HC v1.02 or later (SBC mode)
Board ID: 08DJM-9P63L-DJ3T8-6JKD4-3SJ6K-9A77A
Used output buffers: 1 of 40 (17 max)
=== RTOS ===
Static ram: 155184
Dynamic ram: 87836 of which 4872 recycled
Never used RAM 98100, free system stack 204 words
Tasks:=== Diagnostics ===
RepRapFirmware for Duet 3 MB6HC version 3.5.0-rc.3 (2024-01-24 17:58:49) running on Duet 3 MB6HC v1.02 or later (SBC mode)
Board ID: 08DJM-9P63L-DJ3T8-6JKD4-3SJ6K-9A77A
Used output buffers: 1 of 40 (17 max)
=== RTOS ===
Static ram: 155184
Dynamic ram: 87860 of which 4848 recycled
Never used RAM 98100, free system stack 204 words
Tasks: SBC(2,ready,0.8%,402) HEAT(3,nWait 1,0.0%,323) Move(4,nWait 6,0.0%,336) CanReceiv(6,nWait 1,0.0%,940) CanSender(5,nWait 7,0.0%,334) CanClock(7,delaying,0.0%,334) TMC(4,nWait 6,8.6%,56) MAIN(2,running,90.6%,103) IDLE(0,ready,0.0%,30), total 100.0%
Owned mutexes: HTTP(MAIN)
=== Platform ===
Last reset 01:31:17 ago, cause: software
Last software reset at 2024-02-02 12:19, reason: StuckInSpinLoop, Gcodes spinning, available RAM 98100, slot 2
Software reset code 0x4083 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x0043580f BFAR 0x00000000 SP 0x2041b8b4 Task MAIN Freestk 1983 ok
Stack: 0041a123 0041a13a 210f0000 468cdcfa 410a5e00 b7f83718 3f4d8366 00000000 20439a4c 00000003 00000002 00000002 000013c3 20439a94 00000196 2042fdd0 00000000 43425320 722c3228 74696157 0000003a 00000000 00000000 00000000 0001571c 00001308 00000000
Error status: 0x00
MCU temperature: min 30.1, current 32.0, max 32.2
Supply voltage: min 24.0, current 24.1, max 24.2, under voltage events: 0, over voltage events: 0, power good: yes
12V rail voltage: min 11.7, current 12.2, max 13.0, under voltage events: 0
Heap OK, handles allocated/used 0/0, heap memory allocated/used/recyclable 0/0/0, gc cycles 0
Events: 0 queued, 0 completed
Driver 0: standstill, SG min n/a, mspos 344, reads 5335, writes 22 timeouts 0
Driver 1: standstill, SG min n/a, mspos 1016, reads 5335, writes 22 timeouts 0
Driver 2: standstill, SG min n/a, mspos 680, reads 5336, writes 21 timeouts 0
Driver 3: standstill, SG min n/a, mspos 424, reads 5336, writes 21 timeouts 0
Driver 4: standstill, SG min n/a, mspos 168, reads 5336, writes 21 timeouts 0
Driver 5: standstill, SG min n/a, mspos 4, reads 5339, writes 19 timeouts 0
Date/time: 2024-02-02 13:50:57
Slowest loop: 1.91ms; fastest: 0.09ms
=== Storage ===
Free file entries: 20
SD card 0 not detected, interface speed: 37.5MBytes/sec
SD card longest read time 0.0ms, write time 0.0ms, max retries 0
=== Move ===
DMs created 125, segments created 0, maxWait 0ms, bed compensation in use: none, height map offset 0.000, max steps late 0, ebfmin 0.00, ebfmax 0.00
no step interrupt scheduled
Moves shaped first try 0, on retry 0, too short 0, wrong shape 0, maybepossible 0
=== DDARing 0 ===
Scheduled moves 0, completed 0, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
=== DDARing 1 ===
Scheduled moves 0, completed 0, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
=== Heat ===
Bed heaters 0 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1, chamber heaters -1 -1 -1 -1, ordering errs 0
=== GCodes ===
Movement locks held by null, null
HTTP* is doing "M122" in state(s) 0
Telnet is idle in state(s) 0
File is idle in state(s) 0
USB is idle in state(s) 0
Aux is idle in state(s) 0
Trigger* is idle in state(s) 0
Queue is idle in state(s) 0
LCD is idle in state(s) 0
SBC is idle in state(s) 0
Daemon is idle in state(s) 0
Aux2 is idle in state(s) 0
Autopause is idle in state(s) 0
File2 is idle in state(s) 0
Queue2 is idle in state(s) 0
Q0 segments left 0, axes/extruders owned 0x0000000
Code queue 0 is empty
Q1 segments left 0, axes/extruders owned 0x0000000
Code queue 1 is empty
=== Filament sensors ===
check 0 clear 47911997
Extruder 0: pos 2160.00, errs: frame 0 parity 0 ovrun 0 pol 0 ovdue 0
=== CAN ===
Messages queued 49293, received 0, lost 0, errs 25951542, boc 0
Longest wait 0ms for reply type 0, peak Tx sync delay 0, free buffers 50 (min 50), ts 27387/0/0
Tx timeouts 0,0,27386,0,0,21905 last cancelled message type 30 dest 127
=== SBC interface ===
Transfer state: 5, failed transfers: 0, checksum errors: 0
RX/TX seq numbers: 39880/18235
SPI underruns 0, overruns 0
State: 5, disconnects: 0, timeouts: 0 total, 0 by SBC, IAP RAM available 0x255cc
Buffer RX/TX: 0/0-0, open files: 0
=== Duet Control Server ===
Duet Control Server version 3.5.0-rc.3 (2024-01-29 19:29:19)
Failed to deserialize the following properties:
- MoveSegmentation -> Int32 from 2.0
Code buffer space: 4096
Configured SPI speed: 8000000Hz, TfrRdy pin glitches: 4
Full transfers per second: 39.18, max time between full transfers: 530.4ms, max pin wait times: 36.8ms/8.8ms
Codes per second: 0.02
Maximum length of RX/TX data transfers: 7224/920
```

### Please upload the content of your config.g file.

```gcode
Config.g
```

### Details specific to your printer.

Meltingplot MBL 480

### What happened?

Send M122 via DWC.

Expected result
Get M122 Diagnostics result.

Observed result
Lost connection to Duet.

Steps to reproduce
M122 < keep pressing enter to send consecutive M122 requests as fast as possible until it will stuck>

[image]

dc42, 2024-02-05:
Tim, thanks for your contribution and PR. I have implemented a similar fix in the Duet3Expansion project.

T3P3, 2024-02-07:
back port to 3.4.x

closed 2024-02-07 by T3P3