Duet3D/RepRapFirmware #959: [FeatureRequest]: Allow `M291` to set  `result` to `-1` when blocking modal "Cancel" clicked.

benagricola, 2024-02-15:
### Is your feature request related to a problem? Please describe.

Currently, the documentation for the `result` meta gcode variable states that on clicking the "Cancel" button of a blocking `M291` modal, the `result` value should be set to -1 but this is not the case - the containing input stream is aborted and no code after the `M291` call is evaluated. 

This makes it difficult to give macros the option to abort at a particular location while also allowing some level of recovery from that within the macro (for example to reset speed limits or restart a probing sequence).

### Describe the solution you propose.

As discussed in the forum thread linked below, @dc42 suggested adding a `J2` parameter to `M291` which would cause it to no longer abort if "Cancel" were clicked, but instead set the `result` variable to `-1` as documented.

This would not break existing macros relying on the current `J1` behaviour but would also allow more nuanced usage of `M291` and recovery from cancelling user input.

### Describe alternatives you've considered

For `M291 ... S4` it is possible to add a "Cancel" input option and use the `input` variable to detect if the user clicked "Cancel", but this only works where the input is multiple-option, and cannot be used if the S-value is >= 5 or == 3.

### Provide any additional context or information.

https://forum.duet3d.com/topic/34945/meta-gcode-result-variable-inconsistent-with-docs/4?_=1707983978818

chrishamm, 2024-02-27:
When this is addressed the corresponding implementation in DSF needs to be changed again as well. Unlike in v3.4, in v3.5 `result` is evaluated by RRF which can be off if the result of potential custom G/M-codes diverges from the last result.

T3P3, 2024-05-08:
https://forum.duet3d.com/topic/35663/m291-j2-1/3

Further request

dc42, 2024-08-31:
Related: set result to -1 if the message box times out. See https://forum.duet3d.com/topic/36445/m291-t-value-to-return-a-result/2.

Tinchus2009, 2024-09-27:
Hi. I was directoed to read here after I posted: https://forum.duet3d.com/topic/36694/m291-message-non-blocking-execution/3?_=1727461179293

In J4 cancell the job/macro by pressing CCANCEL, why also cancell the macro/job if the message box is configured to time out? How I see it: timeout means no answer was given, lets continue. If you dont want to continue, you had the CANCEL BUTTOM

dc42, 2024-11-22:
Implemented in 3.6-dev standalone mode by https://github.com/Duet3D/RepRapFirmware/pull/1057. Needs to be tested in SBC mode.

closed 2024-12-04 by T3P3