Duet3D/RepRapFirmware #961: Storing and dereferencing object references in variables

Mogatek, 2024-02-29:
### Is your feature request related to a problem? Please describe.

As per [this forum post](https://forum.duet3d.com/topic/35033/object-reference-in-variable?_=1709050134887), I tried to store a ref to partial sub-tree of the object model in a variable for easier access and more readable code.

```
var tool = 0
var toolHeater = heat.heaters[tools[var.tool].heaters[0]]
 
;--- Accessing the "current" member. As of now, this is not possible:
echo var.toolHeater.current

;--- Expected behaviour would be the equivalent to:
echo heat.heaters[tools[var.tool].heaters[0]].current
```

### Describe the solution you propose.

It would be a great addition to the syntax if we could somehow store object references in variables, then later dereference them as required, accessing the original part of the object model.

### Describe alternatives you've considered

As of now, @dc42 has confirmed that this is not possible. The only way is to use a whole path to the endpoint at every place where it is required.

w1ebr, 2024-02-29:
What about using string substitution (similar to using a preprocessor variable) vs referencing and dereferencing?

dc42, 2024-03-01:
Correct, there is no facility to interpret a string expression as a command.