Duet3D/RepRapFirmware #964: [Bug]: Hangprinter mode (M669 K6) crashes the board and causes a restart

goatchurchprime, 2024-03-05:
### Duet Forum Discussion Thread

n/a

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [X] Other
- [ ] None

### Firmware Version

FIRMWARE_NAME: RepRapFirmware for Duet 3 MB6XD FIRMWARE_VERSION: 3.5.0-rc.3 ELECTRONICS: Duet 3 MB6XD v1.0 FIRMWARE_DATE: 2024-01-24 17:59:29

### Duet Web Control Version

Duet Web Control 3.4.1

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122
=== Diagnostics ===
RepRapFirmware for Duet 3 MB6XD version 3.5.0-rc.3 (2024-01-24 17:59:29) running on Duet 3 MB6XD v1.0 (standalone mode)
Board ID: 08DLM-956DA-M24S4-7JKF0-3S86R-1BMAT
Used output buffers: 1 of 40 (18 max)
=== RTOS ===
Static ram: 153600
Dynamic ram: 117104 of which 0 recycled
Never used RAM 75288, free system stack 207 words
Tasks: NETWORK(1,ready,34.2%,183) ETHERNET(5,nWait 7,0.1%,325) HEAT(3,nWait 6,0.0%,368) Move(4,nWait 6,0.0%,335) CanReceiv(6,nWait 1,0.0%,940) CanSender(5,nWait 7,0.0%,334) CanClock(7,delaying,0.0%,339) MAIN(1,running,65.5%,444) IDLE(0,ready,0.2%,30), total 100.0%
Owned mutexes:
=== Platform ===
Last reset 00:00:57 ago, cause: software
Last software reset at 2024-03-05 12:44, reason: MemoryProtectionFault mmarValid daccViol, none spinning, available RAM 72252, slot 2
Software reset code 0x0173 HFSR 0x00000000 CFSR 0x00000082 ICSR 0x00400804 BFAR 0x00000000 SP 0x20420648 Task NETW Freestk 337 ok
Stack: 0000076a 20428260 0000006a 00000000 004b2866 0041ad5d 0041ad5c 61030000 2000302e 204208d8 00000000 204330f0 20420700 00000036 004a5be0 20420748 004a5aa0 00000000 004a80d8 20431910 204208d8 00000000 20428260 20440f7e 20440f7e 00000005 004a5c34
Error status: 0x00
MCU temperature: min 23.4, current 23.8, max 23.9
Supply voltage: min 12.0, current 12.1, max 12.1, under voltage events: 0, over voltage events: 0, power good: yes
12V rail voltage: min 11.4, current 11.4, max 11.5, under voltage events: 0
Heap OK, handles allocated/used 0/0, heap memory allocated/used/recyclable 0/0/0, gc cycles 0
Events: 0 queued, 0 completed
Driver 0: ok
Driver 1: ok
Driver 2: ok
Driver 3: ok
Driver 4: ok
Driver 5: ok
Date/time: 2024-03-05 12:45:47
Slowest loop: 4.52ms; fastest: 0.07ms
=== Storage ===
Free file entries: 20
SD card 0 detected, interface speed: 25.0MBytes/sec
SD card longest read time 2.3ms, write time 0.0ms, max retries 0
=== Move ===
DMs created 125, segments created 0, maxWait 0ms, bed compensation in use: none, height map offset 0.000, max steps late 0, ebfmin 0.00, ebfmax 0.00
no step interrupt scheduled
Moves shaped first try 0, on retry 0, too short 0, wrong shape 0, maybepossible 0
=== DDARing 0 ===
Scheduled moves 0, completed 0, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
=== DDARing 1 ===
Scheduled moves 0, completed 0, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
=== Heat ===
Bed heaters -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1 -1, chamber heaters -1 -1 -1 -1, ordering errs 0
=== GCodes ===
Movement locks held by null, null
HTTP is idle in state(s) 0
Telnet is idle in state(s) 0
File is idle in state(s) 0
USB is idle in state(s) 0
Aux is idle in state(s) 0
Trigger is idle in state(s) 0
Queue is idle in state(s) 0
LCD is idle in state(s) 0
SBC is idle in state(s) 0
Daemon is idle in state(s) 0
Aux2 is idle in state(s) 0
Autopause is idle in state(s) 0
File2 is idle in state(s) 0
Queue2 is idle in state(s) 0
Q0 segments left 0, axes/extruders owned 0x0000000
Code queue 0 is empty
Q1 segments left 0, axes/extruders owned 0x0000000
Code queue 1 is empty
=== CAN ===
Messages queued 290, received 0, lost 0, errs 275239, boc 0
Longest wait 0ms for reply type 0, peak Tx sync delay 0, free buffers 50 (min 50), ts 290/0/0
Tx timeouts 0,0,289,0,0,0 last cancelled message type 30 dest 127
=== Network ===
Slowest loop: 6.83ms; fastest: 0.03ms
Responder states: MQTT(0) HTTP(0) HTTP(0) HTTP(0) HTTP(0) HTTP(0) HTTP(0) FTP(0) Telnet(0) Telnet(0)
HTTP sessions: 2 of 8
= Ethernet =
Interface state: active
Error counts: 0 0 0 1 0 0
Socket states: 5 2 2 2 2 0 0 0
=== Multicast handler ===
Responder is inactive, messages received 0, responses 0
```

### Please upload the content of your config.g file.

```gcode
M550 P"duet3d"             ; set printer name
M552 S1               ; enable network and acquire dynamic address via DHCP
```

### Details specific to your printer.

Servo motor controller for CNC

### What happened?

If you execute "M669 K6" on the Console the board crashes, disconnects and eventually reboots. 

All other kinematics types work okay and either set to the defaults or give an error: 

eg
```
> M669 K7
> M669
Kinematics is Polar, 100 segments/sec, min. segment length 0.20mm, radius 0.0 to 150.0mm, homed radius 0.0mm, max table acc. 30.0deg/sec^2, max table speed 30.0deg/sec

> M669 K9
Error: M669: missing parameter 'X'
```

goatchurchprime, 2024-03-05:
We just compiled from sources and uploaded, and it works for the following firmware version:

FIRMWARE_NAME: RepRapFirmware for Duet 3 MB6XD FIRMWARE_VERSION: 3.4.6+ ELECTRONICS: Duet 3 MB6XD v1.0 FIRMWARE_DATE: 2024-03-05 13:14:27

from this commit:
https://github.com/paTelshREya/RepRapFirmware/commit/9409dafb957e2ccc61c928bed42ebe2c8460636b

Now it works correctly:

```
> M669 K6
> M669
Kinematics is Hangprinter, 100 segments/sec, min. segment length 0.20mm
A:0.00, -2000.00, -100.00
B:2000.00, 1000.00, -100.00
C:-2000.00, 1000.00, -100.00
D:0.00, 0.00, 3000.00
P:Print radius: 1500.0
```

dc42, 2024-03-06:
Now fixed in 3.5-mms-changes branch. Will be merged into 3.5-dev in due course.

dc42, 2024-03-15:
3.5-mms-changes now merged into 3.5-dev.

closed 2024-03-15 by dc42