Duet3D/RepRapFirmware #967: [Bug]: G10 retract z-hop ignores homing and machine limits

timschneider, 2024-03-13:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/35012/feature-or-bug-3-5-0-rc-3-g10-retract-z-hop-ignores-homing

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [X] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-6XD
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.5.0-rc.3

### Duet Web Control Version

DWC 3.5.0-rc.3

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [X] Yes I use a SBC.
- [ ] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
see thread
```

### What happened?

The printer is unhomed and a retration z-hop is configured via M207.

Steps to reproduce
execute ```G10```

Expected result
The printer should responde with an error, e.g. ```Error: G10: insufficient axes homed```

Observed result
The Z-axis is performing the z-hop ignoring the homing state and machine limit

another apsect of it occures when using cancle

Steps to reproduce
see thread

Observed result

-     start jobfile.g
-     pause printjob via DWC (M24)
-     cancel printjob via DWC (M0)
-     repeat the above steps and the z-axis will accumulate the z-hops, 0.0, 0.5, 1.0, 1.5 ... and so on with an configured z-hop of 0.5mm

Expected result
the retraction state should be preserved by M0 and the z-axis should not accumulate the z-hops.

T3P3, 2024-05-08:
This is two separate issues:

1) G10 with Z hop causing movement when the Z axis is not homed, or causes a move outside of machine limits. Proposed solution will not move when Z axis is not homed, or if the move will move outside of machine limits.
2) G10 state in pause/cancel needs to be documented correct (i.e G10 in pause, G11 in resume, G11 in cancel being the valid way to use firmware retraction).

Also if Z is ever flagged an un homed it should also be flagged as not hopped.

dc42, 2024-05-22:
Fixes tested and found to be working today. Z hop is no longer performed when the Z axis has not been homed. Otherwise, the amount of Z hop is limited according to the maximum reachable Z (which for a linear delta machine depends on the XY position). G11 when Z has been homed undoes the amount of Z hop last performed by G10 on that tool.

closed 2024-05-22 by dc42

timschneider, 2024-05-22:
@dc42 it looks like a very clean fix and well thought out rework of the whole z-hop part. thank you! I'll test it as soon as it is available in the unstable branche.