Duet3D/RepRapFirmware #971: [Bug]: Jerk & acceleration value change after Resume APF

ButalaRishita, 2024-04-06:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/35334/jerk-acceleration-value-change-after-resume-apf?_=1712377194339

### Which Duet products are you using?

- [X] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

RRF 3.5.0 rc3

### Duet Web Control Version

DWC3.2.2

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122 Report
```

### Please upload the content of your config.g file.

```gcode
Config.g
```

### What happened?

I am using Duet 2 wifi with RRF 3.5.0 rc3.

sudden change in jerk and acceleration after resume APF, it's set random value which is not in config and not in gcode.

gcode in which jerk is 300 mm/sec & acceleration 500 mm/s2, when i run the power failure sequence and after resume the print there is sudden vibration in X Y axis, so i am rum M204 to check it's acceleration and the result of it is 50000 mm/s2 and jerk is 600mm/s but gcode generated on jerk 300mm/s and acceleration 500 mm/s2.

dc42, 2024-04-07:
Restoration of M204 settings is now implemented in 3.5.0-rc.4. Restoration of jerk settings has not been done because it is not clear that it is a good idea. The original reason for jerk settings was to specify the limits that the machine is capable of. Unfortunately some slicers now generate commands to set jerk limits. One possibility is in future to treat M566 as setting machine limits (just as M201 sets machine acceleration limits) but treat M205 as setting possibly lower limits for the current print only (just as M204 sets travel and printing acceleration limits for the current print only).

T3P3, 2025-01-19:
Now that the following change has been made to Jerk in 3.6beta2 the restoration of M205 settings will be safer as they are not permitted to be higher than the M566 settings in the machine config.g.

From the [release notes:](https://github.com/Duet3D/RepRapFirmware/wiki/Changelog-RRF-3.x-Beta#reprapfirmware-360-beta2)

> In previous firmware versions, M566 and M205 both adjusted a single set of jerk limits with the only difference being the units that were provided (mm/min for M566 and mm/sec for M205). In this release, RRF maintains separate machine jerk limits and jerk limits for the current job. M566 sets both jerk limits, whereas M205 sets only the jerk limits for the current job. The current job jerk limits are constrained to be no higher than the machine jerk limits. This allows slicers to user M205 to change the allowed jerk without exceeding machine limits.

dc42, 2025-01-20:
Fixed in 3.6-dev branch.

closed 2025-01-20 by dc42