Duet3D/RepRapFirmware #974: [Bug]: G28 X causes Z to crash into bed after X is homed

NNygren, 2024-04-11:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/35418/3-5-0-rc-4-z-print-head-crashes-into-bed-at-end-of-print

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [X] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.5.0-rc.4 

### Duet Web Control Version

3.5.0-rc.4 

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122
=== Diagnostics ===
RepRapFirmware for Duet 3 Mini 5+ version 3.5.0-rc.4 (2024-04-09 09:45:40) running on Duet 3 Mini5plus WiFi (standalone mode)
Board ID: AATGZ-U396U-D65J0-40KM4-NM03Z-HUQXW
Used output buffers: 17 of 40 (18 max)
=== RTOS ===
Static ram: 103232
Dynamic ram: 121344 of which 0 recycled
Never used RAM 16840, free system stack 208 words
Tasks: NETWORK(2,nWait 7,9.9%,241) HEAT(3,nWait 6,0.0%,351) Move(4,nWait 6,0.0%,355) CanReceiv(6,nWait 1,0.0%,940) CanSender(5,nWait 7,0.0%,336) CanClock(7,delaying,0.0%,336) TMC(4,nWait 6,0.8%,111) MAIN(1,running,87.3%,665) IDLE(0,ready,1.1%,30) AIN(4,delaying,0.8%,265), total 100.0%
Owned mutexes: WiFi(NETWORK)
=== Platform ===
Last reset 00:00:19 ago, cause: software
Last software reset at 2024-04-10 22:46, reason: User, Gcodes spinning, available RAM 12780, slot 0
Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x00000000 BFAR 0xe000ed38 SP 0x00000000 Task MAIN Freestk 0 n/a
Error status: 0x00
MCU revision 3, ADC conversions started 19801, completed 19801, timed out 0, errs 0
MCU temperature: min 33.4, current 33.6, max 34.3
Supply voltage: min 24.2, current 24.3, max 24.3, under voltage events: 0, over voltage events: 0, power good: yes
Heap OK, handles allocated/used 0/0, heap memory allocated/used/recyclable 0/0/0, gc cycles 0
Events: 0 queued, 0 completed
Driver 0: standstill, SG min 232, read errors 0, write errors 1, ifcnt 62, reads 1019, writes 10, timeouts 0, DMA errors 0, CC errors 0
Driver 1: standstill, SG min 40, read errors 0, write errors 1, ifcnt 58, reads 1019, writes 10, timeouts 0, DMA errors 0, CC errors 0
Driver 2: standstill, SG min 48, read errors 0, write errors 1, ifcnt 57, reads 1019, writes 10, timeouts 0, DMA errors 0, CC errors 0
[... 176 lines cut ...]
M207 S0.65 F2400 Z0  ; firmware retraction
M593 P"zvddd" F61    ; use input shaping to cancel ringing
```

### Details specific to your printer.

Cartesian

### What happened?

Expected result
At end of print, my GCODE is:
...
G28 X
...
G90 ;absolute positioning
G1 Y200 ;move Y forward to present print

Observed result
Console displayed "Error: G1: intermediate position outside machine limits
Finished printing file 0:/gcodes/Shooter-Disc.gcode, print time was 0h 10m"
G1 Y200 was never executed.

Steps to reproduce
Type in G28 X into DWC command line produced same result

NNygren, 2024-04-11:
The homex.g file generated by the RepRapFirmware Config tool ends with `G1 H2 Z-2 F6000 ; lower Z again` and commenting that out stopped the print head from crashing into the bed after homing.

closed 2024-04-11 by NNygren

dc42, 2024-04-12:
Did you forget to include G90/G91 around those lines? Assuming the homex.g file raised Z by 2mm earlier in the file.