Duet3D/RepRapFirmware #977: [Bug]: Low Frequency on PWM pins with version 3.5.0

Pourya-3deo, 2024-04-18:
### Duet Forum Discussion Thread

I tried I couldn't register

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [X] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

RRF 3.5.0

### Duet Web Control Version

DWC 3.4.6

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [X] Yes I use a SBC.
- [ ] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122 Report
FIRMWARE_NAME: RepRapFirmware for Duet 3 MB6HC FIRMWARE_VERSION: 3.5.0 ELECTRONICS: Duet 3 MB6HC v1.02 or later FIRMWARE_DATE: 2024-04-16 09:43:06
```

### Please upload the content of your config.g file.

```gcode
Config.g

M950 P10 C"io4.out" Q10
```

### Please upload the content of any other releveant macro files.

```gcode
M42 P10 S0.2
```

### Details specific to your printer.

Our own 3d printer

### What happened?

Expected result
Frequency should be 10 HZ

Observed result
Frequency was 17.88 HZ

Steps to reproduce

closed 2024-04-19 by dc42

Pourya-3deo, 2024-04-19:
> This is a known limitation, present because the timer/counter channels on
> the SAME70 processor are only 16 bits wide and the range of input clocks
> available is limited. The timer channels currently use a 150MHz clock
> divided by 128, so the minimum available PWM frequency is
> 150MHz/(128*65535) = 17.88Hz. We may be able to reduce the minimum
> frequency to 11.44Hz by switching to a different clock.
>
> The PWM-capable outputs on the 6HC that are drive by timer/counter
> channels on the 6HC are: OUT4 OUT5 OUT8 OUT7 IO4.OUT IO5.OUT IO7.OUT
>
> The PWM-capable outputs on the 6HC that are driven by the PWM peripheral
> and have a lower minimum frequency are: OUT0 OUT1 OUT2 OUT3 OUT6 OUT9

Thank you for your email, I was worried about that and hoped for a
solution, is there any software based PWM with a macro maybe?

dc42, 2024-04-19:
If you need a lower PWM frequency then use one of the outputs that is controlled by the PWM peripheral, which I listed in my reply above.

Pourya-3deo, 2024-04-19:
So I tried OUT0 and OUT9 but I don't get any accurate square shaped signal. Here is the code:
Config:
M950 P10 C"out9" Q10
Macro:
M42 P10 S0.5