Duet3D/RepRapFirmware #979: [FeatureRequest]: allow more than one keep-out zone

Braintoe, 2024-04-20:
### Is your feature request related to a problem? Please describe.

Currently, only one keep-out zone can be defined. The feature could be very helpful to avoid print head crashes with e.g. glass plate clamps on an FFF printer, but usually there is more than one of those, which severely impacts the use-cases of the M599 function.

### Describe the solution you propose.

For discussion, see https://forum.duet3d.com/topic/35507/multiple-keep-out-zones-m599

### Describe alternatives you've considered

Apart from taking care when moving the printer, I see no alternatives at the moment.

### Provide any additional context or information.

Ex. Photos, mockups, etc.

Braintoe, 2024-12-01:
@dc42 out of curiosity: since 3.6.0 is already in beta, I guess that enhancement won't make it into this version, correct? I will hope for 3.7.0 then :-)

T3P3, 2026-09-16:
Forum thread: https://forum.duet3d.com/topic/31974