Duet3D/RepRapFirmware #981: Feature: MODBUS RTU support

dc42, 2024-04-22:
Add support for MODBUS RTU using any free Aux port in conjunction with a UART to RS485 converter and an IO_OUT pin to control the transmitter.

dc42, 2024-04-22:
I have already reserved additional parameters to allow Mm260 and M261 to be used for thus, see https://reprap.org/wiki/G-code#M260:_i2c_or_Modbus_Send_Data.

developeralgo8888, 2024-06-24:
Why can't the dev team add RS485 support? It critical in high speed automation to any third party devices and reliable. I was really surprised it was requested for RRF 3.6.0 and now it has been shifted to "After 3.6.0". Its disappointing.

dc42, 2024-08-10:
Modbus RTU support has been added to the 3.6-dev branch of RRF and is available in recent 3.6.0-alpha builds. See the new M260.1 and M261.1 commands. The V parameter of M261.1 is not yet implemented.

dc42, 2024-08-16:
Implementation in 3.6-dev completed and tested by user.

closed 2024-08-16 by dc42