Duet3D/RepRapFirmware #986: [FeatureRequest]: Have isnan return true for an array value instead of error

DonStauffer, 2024-04-26:
### Is your feature request related to a problem? Please describe.

I'd like to be able to tell whether something is an array or not.

### Describe the solution you propose.

I propose having the isnan function return true for an array value

### Describe alternatives you've considered

Everything I've tried throws an error either when the value is or is not an array. For example, passing the first element of an array to exists produces an error (logically) if the value is actually not an array.

### Provide any additional context or information.

What I'm doing right now is trying to pass a parameter to a macro, where the macro behaves differently if you pass a 1D or 2D array.

dc42, 2025-09-15:
I'm closing this issue because we're not going to implement nit. Feel free to add comments if you have suggestions for alternative mechanisms to determine whether a variable is an array, or more generally to determine the type of a variable.

adambx, 2025-09-16:
Suggesting a function called type() that would output a string type identifier like `int`, `float`, `arr/array`, `obj/object`, `str/string` `null?`. 
Of course this would add some overhead if you want to check types during runtime `if type(var.variable) == "array"`, but it would add the benefit of echoing types directly with `echo type(var.variable)` instead of 
```
if isArray(var.variable)
    echo "array"

if isString(var.variable)
    echo "string"
```
depends on the use case.