Duet3D/RepRapFirmware #990: [bug]: DoLookahead modifies extrusion move followed by z move

gloomyandy, 2024-04-29:
Possible problem identified in RRF 3.5.1 and this report created as requested by DC42

While investigating another issue I noticed that the following code:
G1 E-2 F200
G1 Z200 F1000

Results in DoLookahead modifying the end speed of the extrusion move from zero to allow the Z move to start at a higher speed. This is probably not correct as the two moves are in effect not related. The following code results in the moves operating as expected:
G1 E-2 F200
M400
G1 Z200 F1000

dc42, 2024-04-29:
Proposed fix:
At DDA.cpp 1175 change:
`if (startSpeed + acceleration * MinimumAccelOrDecelClocks > requestedSpeed)`
to:
`if (startSpeed + acceleration * MinimumAccelOrDecelClocks > requestedSpeed && startSpeed >= requestedSpeed * 0.95)`
and similarly with endSpeed in line 1187. So only reduce topSpeed to avoid the very short segment if the start or end speed is almost as high as the requested speed anyway.

dc42, 2024-05-22:
Confirmed fixed by Andy S.