Duet3D/RepRapFirmware #991: Better support for changing the CAN bit rate

dc42, 2024-05-02:
We could support a small number of lower CAN bit rates (e.g. 500kbits/sec and 250kbits/sec) better if we changed the bootloader code to try listening for clock sync messages at various bit rates; and when one is received, if the bit rate doesn't match the bit rate configured in NVRAM, update the NVRAM. We would need a separate LED flash code for "no clock message received".

T3P3, 2025-02-12:
related to https://github.com/Duet3D/RepRapFirmware/issues/1085

dc42, 2025-04-30:
Plan for implementing this and related features:
1. Time sync messages shall always be sent without using BRS.
2. Use two of the spare bits in the time sync message to indicate whether BRS should be used for most or all other message types and whether the BRS multiplier is 2, 3 or 4.
3. [Breaking change] Add new "Enable CAN" GCode command and require it to be used in config.g before referring to any CAN-connected expansion boards. We could use M954 without the A parameter for this. Using this command will replace the current requirement to use a G4 delay command in config.g before the first reference to a CAN-connected board. This command will include parameters to specify what bit rate should be used (out of a set of standard bit rates) and whether BRS should be used. Until this command is used, no CAN messages (including time sync messages) shall be transmitted.
4. [Optional] After the Enable CAN command is used, as well as starting transmission of time sync messages, the main board firmware shall delay to allow expansion boards to start up and establish the bit rate, then it will send a message asking all expansion board to announce themselves.
5. When trying to install firmware, the expansion bootloader firmware will attempt to establish what bit rate is being used by listening for time sync messages at each of the standard bit rates. If necessary it will update the bit rate stored in NVRAM. The expansion board main firmware can use the data in NVRAM to avoid having to listen for time sync messages at different bit rates, because such scanning could significantly delay expansion board startup.
6. The procedure for changing the main bit rate will be to change the bit rate in the M954 command in the main board config.g file, then execute a CAN reset on all expansion boards in order to invoke the bootloader to scan for the new bit rate.
7. To change the BRS multiplier it will only be necessary to update the BRS multiplier in the M954 command in the main board config.g file, then power the whole system off and on again. Alternatively it may be sufficient to use M999 to reset the whole system.
8. We need to consider which messages should not use BRS for resilience, e.g. Emergency Stop messages.